IMRCLab/crazyswarm2 #5: Running console on windows

knmcguire, 2021-11-10:
Hi! lets start a new thread about the console now that the build is solved.

I tried to run the following in ROS2 in windows 10.

`ros2 run crazyswarm2 console --uri radio://0/70/2M/E7E7E7E703`

but the program immediately closes, and /rosout is not detecting any activity. I've tried this on both libusb-win and winUSB libraries on zadig for the crazyradio.

knmcguire, 2021-11-11:
Just a quick references, on ubuntu it works, as I see this after connection:

```
$ ros2 run crazyswarm2 console --uri radio://0/70/2M/E7E7E7E703

SYS: ----------------------------
SYS: Crazyflie 2.1 is up and running!
SYS: Build 326:15e4e3f42410 (2021.03 +326) MODIFIED
SYS: I am 0x2030374357535017004F001E and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
DECK_CORE: 0 deck(s) found
IMU: BMI088: Using I2C interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Complementary (1) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushed motor driver
SYS: About to run tests in system.c.
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
SYS: Self test passed!
STAB: Wait for sensor calibration...
SYS: Free heap: 20552 bytes
STAB: Ready to fly.
```

whoenig, 2021-11-15:
It would be helpful to know if https://github.com/bitcraze/crazyflie-link-cpp works for you (there is an `example_console` application if you build the repo, which has the same functionality).

knmcguire, 2021-11-16:
I was able to build that example, but I get the following error when I run example_console in debug mode in visual studio 2019

```
Exception thrown: read access violation.
dev_handle was 0xFFFFFFFFFFFFFFBF.
```

or 

```
Unhandled exception thrown: read access violation.
**dev_handle** was 0xFFFFFFFFFFFFFFBF.
```

This is both with winUSB and libusb-win32

knmcguire, 2021-11-17:
I guess we just have to wait until https://github.com/bitcraze/crazyflie-link-cpp/issues/13 is resolved, which is related to this PR in the libusb library... https://github.com/libusb/libusb/pull/1016. So maybe if we build the library from source?

It is strange that it also fails with winusb as well... perhaps @ataffanel might be able to give us some pointers?

whoenig, 2021-11-17:
On Windows, we already build libusb ourselves (it's a submodule), so we are free to pick up any recent changes if needed. Mac/Linux use the locally installed version of the library. Perhaps you could pull the latest master and see if it's still a problem?

knmcguire, 2021-11-17:
Ah yeah you are right! I totally missed the submodule. thanks for notifying :)

I'll try it out these days. Currently confined to Ubuntu for the time being.

whoenig, 2021-11-17:
Yeah, I have the same problem. I guess I would need a dedicated machine just for Windows development...

knmcguire, 2021-11-23:
Seems that we are one step closer to crazyswarm on windows: https://github.com/bitcraze/crazyflie-link-cpp/pull/22

The libusb submodule update worked!

knmcguire, 2021-11-23:
Hmmm... so I updated crazyflie_link_cpp... but ros2 run console still immediately closes, so it might just be an ros problem instead. The crazyflie_link_cpp console works so let's see

whoenig, 2021-11-23:
The ROS version has crazyflie_tools -> crazyflie_cpp -> crazyflie-link-cpp as components, so the bug could also be in crazyflie_cpp or crazyflie_tools. I did test this function last Friday on Ubuntu and it worked (with ROS2).

knmcguire, 2021-11-23:
Yeah, it's submodule inception 😅 it also worked for me on ubuntu indeed.

knmcguire, 2021-11-23:
so it seems to be in crazyflie_tools still. With debugging the console program I get this:
'console.exe' (Win32): Loaded 'C:\Users\kimbe\Documents\Development\C\crazyflie_tools\build\Debug\console.exe'. Symbols loaded.
'console.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. 
'console.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. 
'console.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. 
The thread 0x3780 has exited with code -1073741515 (0xc0000135).
The thread 0x1fb0 has exited with code -1073741515 (0xc0000135).
The program '[25000] console.exe' has exited with code -1073741515 (0xc0000135) 'A dependent dll was not found'.

And this error message:

[image]

Something still is off with the boost, so let's see if we have to adjust the cmake in that repo as well.

knmcguire, 2021-11-23:
ah that was because the boost library was not in the environment path.... not sure if this should be a user instruction or not. Seems messy!

whoenig, 2021-11-23:
Might be easiest to get rid of boost, to be honest. https://github.com/CLIUtils/CLI11 seems pretty popular and is header-only for program_options. Otherwise, static linking helps with the mess of missing dlls, but I think you had problems with that earlier (mismatching compiler versions).

knmcguire, 2021-11-23:
so good news, crazyflie_tools repo with the new crazy_link_cpp subsubmodule works! 

My submodules in my crazyswarm2 was a bit messed up so I did the unholy think of deleting the crazyflie_tools, copying my local one that I managed to work and tada:
```
C:\Users\kimbe\Documents\Development\ROS\ros2_ws>ros2 run crazyswarm2 console
SYS: ----------------------------
SYS: Crazyflie 2.1 is up and running!
SYS: Production release 2021.06
SYS: I am 0x203937434848501700DECK_INFO: Reading deck nr:0 [FAILED]. No driver will be initialized!
DECK_CORE: 0 deck(s) found
IMU: BMI088: Using I2C interface.
IMU: BMI088 Gyro connection [OK].
IMU: BMI088 Accel connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using Complementary (1) estimator
CONTROLLER: Using PID (1) controller
MTR-DRV: Using brushed motor driver
SYS: About to run tests in system.c.
EEPROM: I2C connection [OK].
STORAGE: Storage check [OK].
IMU: BMI088 gyro self-test [OK]
SYS: Self test passed!
STAB: Wait for sensor calibration...
SYS: Free heap: 20560 bytes
STAB: Ready to fly.
```

So I think it is just a matter of updating the submodules with the latest changes

knmcguire, 2021-11-23:
Ah sorry, didn't see your comment before. Eventhough now the path solution worked for me, Yeah I think indeed that is best to get rid of boost, but let's make a seperate issue about that after we get some kind of working version for windows.

knmcguire, 2021-11-23:
btw, i started with a PR to start the chain of submodule changes, but it became a bit too complicated in terms of which branches to use and I don't want to mess things up. Shall I just leave it up to you @whoenig to update these submodules then?

whoenig, 2021-11-24:
Boost: I added a github issue to track this.
Submodules: Sorry for the mess. Let's try to get my custom changes of crazyflie-link-cpp in the master branch, and then I can recursively update.

knmcguire, 2021-11-24:
ah yeah no worries! It would be indeed good to start integrating those back to master if they are considered more stable now. I think also that probably within Bitcraze we should consider to reignitethe efforts crazyflie-cpp-link integrated in the python lib again.

knmcguire, 2021-12-06:
so with the latest submodules and everything, I gave it another try while I'm back in the Windows realm, and viola! the console example works now from windows 😄 so that means that the basic functionality works. 

Haven't tried the other examples yet but that is for another issue in the future, as this one was just specifically for the console and the crazyradio basic connection. I'll close it!

closed 2021-12-06 by knmcguire