IMRCLab/crazyswarm2 #6: Add support for cmd_vel teleoperation

whoenig, 2021-11-26:
In Crazyswarm(1), no teleoperation (control using a joystick) was ever implemented. Now that Crazyflie_ros is deprecated, this feature should be part of Crazyswarm2.

Steps:
1. Update the teleop node to publish cmd_vel (feature to be enabled using a parameter). Examples are https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/src/quadrotor_teleop.cpp and https://github.com/ethz-asl/rotors_simulator/blob/master/rotors_joy_interface/src/joy.cpp
2. Update the crazyswarm_server to subscribe to cmd_vel and to send the appropriate command to the Crazyflie, see https://github.com/whoenig/crazyflie_ros/blob/df3ce76f700953cfa2794b8fd4628ea6b93c92c4/crazyflie_driver/src/crazyflie_server.cpp#L310-L322

knmcguire, 2021-12-06:
Interesting! I was working on doing the gazebo plugin to do the same for the simulator. Perhaps I can take a look at this. 

I guess the idea is to actually make an separate package out of this right?

whoenig, 2021-12-06:
I actually created this one for my PhD student to learn a bit more about ROS2 and she already started:-)

closed 2022-01-14 by whoenig