IMRCLab/crazyswarm2 #17: Colcon Build on Linux Failing due to PCL configuration file not existent

longbow33, 2022-02-04:
Hello Hoenig,
i am trying to build the crazyswarm 2 and it gives the following output:

[1.021s] WARNING:colcon.colcon_core.verb:Some selected packages are already built in one or more underlay workspaces:
	'crazyswarm2_interfaces' is in: /home/sebastian/ros2_ws/install/crazyswarm2_interfaces
	'motion_capture_tracking_interfaces' is in: /home/sebastian/ros2_ws/install/motion_capture_tracking_interfaces
	'motion_capture_tracking' is in: /home/sebastian/ros2_ws/install/motion_capture_tracking
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding crazyswarm2_interfaces motion_capture_tracking motion_capture_tracking_interfaces

This may be promoted to an error in a future release of colcon-core.
Starting >>> crazyswarm2_interfaces
Starting >>> motion_capture_tracking_interfaces
Finished <<< motion_capture_tracking_interfaces [0.70s]
Starting >>> motion_capture_tracking
Finished <<< crazyswarm2_interfaces [1.19s]
Starting >>> py_crazyswarm2
Starting >>> crazyswarm2             
--- stderr: motion_capture_tracking  
CMake Error at CMakeLists.txt:16 (find_package):
  By not providing "FindPCL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PCL", but
  CMake did not find one.

  Could not find a package configuration file provided by "PCL" with any of
  the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< motion_capture_tracking [2.60s, exited with code 1]
Aborted  <<< py_crazyswarm2 [2.59s]                                     
Aborted  <<< crazyswarm2 [24.5s]                                 
                                 
Summary: 2 packages finished [26.7s]
  1 package failed: motion_capture_tracking
  2 packages aborted: crazyswarm2 py_crazyswarm2
  3 packages had stderr output: crazyswarm2 motion_capture_tracking py_crazyswarm2
  1 package not processed

Any idea, what could fix that ?

whoenig, 2022-02-04:
You need to install PCL, perhaps with `libpcl-dev`.

For the warning at the beginning, make sure you do not re-use the same shell for colcon and running your scripts.

longbow33, 2022-02-04:
After installing libpcl-dev it throws this error:

Starting >>> crazyswarm2_interfaces
Starting >>> motion_capture_tracking_interfaces
Finished <<< motion_capture_tracking_interfaces [0.70s]
Starting >>> motion_capture_tracking
--- stderr: motion_capture_tracking  
CMake Error at /home/sebastian/ros2_foxy/ros2-linux/share/rmw_implementation/cmake/rmw_implementation-extras.cmake:54 (message):
  The RMW implementation has been specified as 'rmw_fastrtps_cpp' via the
  default when rmw_implementation was built, but it is not available at this
  time.

  

  Currently available middlewares:

  ''
Call Stack (most recent call first):
  /home/sebastian/ros2_foxy/ros2-linux/share/rmw_implementation/cmake/rmw_implementationConfig.cmake:41 (include)
  /home/sebastian/ros2_foxy/ros2-linux/share/rcl/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/sebastian/ros2_foxy/ros2-linux/share/rcl/cmake/rclConfig.cmake:41 (include)
  /home/sebastian/ros2_foxy/ros2-linux/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/sebastian/ros2_foxy/ros2-linux/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /home/sebastian/ros2_foxy/ros2-linux/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/sebastian/ros2_foxy/ros2-linux/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  CMakeLists.txt:10 (find_package)

---
Failed   <<< motion_capture_tracking [0.24s, exited with code 1]
Aborted  <<< crazyswarm2_interfaces [1.04s]                             

Summary: 1 package finished [1.99s]
  1 package failed: motion_capture_tracking
  1 package aborted: crazyswarm2_interfaces
  1 package had stderr output: motion_capture_tracking
  3 packages not processed

whoenig, 2022-02-04:
We only support ROS2 Galactic at the moment.

closed 2022-02-04 by longbow33