IMRCLab/crazyswarm2 #38: Automatic emergency if a drone can't be tracked for a specified period of time

whoenig, 2022-06-10:
If we lose tracking (e.g., via a motion capture system), the UAV typically keeps flying until it is crashed. Depending on the crash, the motors might keep spinning (e.g., if the UAV is caught in a net).
Desired behavior: have a configuration option to send an emergency shutoff automatically in this case.

varadVaidya, 2024-12-03:
I might be implementing something similar to this for my work as well. I am currently thinking of using the emergency service to shutoff all motors after 1.5 secs of no tracking. I feel that the both tracking methods i.e. single marker tracking, and rigid body tracking with MoCap vendor software, should support the automatic emergency.

Let me know what are your thoughts on this. I think I can take up on implementing and testing this out.

knmcguire, 2024-12-03:
We should also be aware of the supervisor in the Crazyflie firmware: https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/functional-areas/supervisor/. 

There is currently nothing blocking take off if the estimate is bad, but we had a master student at the office once that wrote a thesis about possible ways to implement this. Perhaps interesting to take a look at as well:

https://lup.lub.lu.se/student-papers/search/publication/9175409