IMRCLab/crazyswarm2 #41: The use of --symlink-install can be unpredictable

whoenig, 2022-06-14:
Some of our packages use Python with setup.py. In that case, symlink-install doesn't work as intended.

We should either switch to cmake-based build everywhere (even for Python-only packages), or discourage the use of --symlink-install.

See: https://github.com/colcon/colcon-core/issues/407

whoenig, 2022-08-09:
This is currently only relevant for Python packages that use config and/or launch files. At the moment, this is only the experimental crazyflie_server_py package.

knmcguire, 2022-08-09:
until now I haven't had many issues with the --symlink-install myself when working on the crazyflie_server_py.  I do remember that I indeed had issues with the config yaml files a while ago, but I've recently updated my galactic ROS2 and haven't seen this issue anymore actually. Maybe it's secretively fixed?

whoenig, 2022-08-09:
If you don't see a problem, then we can close this issue for now.

closed 2022-08-09 by whoenig

HappySamuel, 2024-02-02:
Hi @whoenig , @knmcguire 

I am using the python package, i faced the issue that when is use `colcon build --symlink-install`, but no symlink created at all. How shall i solve this?

Best,
Samuel

whoenig, 2024-02-02:
@HappySamuel, please open a new discussion and describe what problem you are facing (what did you do, what was the observed behavior, what is the expected behavior?). Thanks!