IMRCLab/crazyswarm2 #44: crazyflie_server python cflib edition

knmcguire, 2022-07-19:
So I want to contribute a cflib backend crazyflie server node, not to replace the ccp version but to give the users more flexibility to choose which one.

These are the first features that I will implement (per PR)
- [ ] 1. A simple crazyflie server node for teleoperation (crazyflie_server_py see #43)
- [ ] 2. Extend that with logging
- [ ] 3. Extend that with an tf output of the state estimation.

I'll update this list soon, as I want do some read-up first and look through the current server and crazyswarm1 in more detail. But at  just for reference that it is in the pipeline and that I connect the PRs to this grander issue/ticket

knmcguire, 2022-07-19:
I think I will first start with the node for just one cf first.

knmcguire, 2022-07-19:
I've started an branch on my fork of crazyswarm2: https://github.com/knmcguire/crazyswarm2/tree/add_py_server. I could already make an PR and put it to draft so that the progress can be seen, but perhaps that will also create a lot of message noise perhaps?

knmcguire, 2022-08-01:
Just an heads up! I'm working now on the parameter and logging handling on this branch: https://github.com/knmcguire/crazyswarm2/tree/py_server_log_param. Currently at handling parameters from crazyswarm2_server.yaml.

There is this new fully_connected callback possibility that is enabled when all the parameters are downloaded and all tocs are received, so that is possibly to keep in mind when the new release comes out.

whoenig, 2022-08-02:
sounds great! I started working on the renaming, hopefully I can create a PR later today.

knmcguire, 2022-08-02:
Actually.. it seems that I'll implement the fully connected functionality afterall... especially the parameter handling is quite tedious without. But I guess at this stage of development it is reasonable to say to update cflib from source 🙂

knmcguire, 2022-08-09:
@whoenig sorry for the delay. I've run into some threading handling issues with cflib <-> ros2 node that makes it difficult to check if parameters are correctly set in the cf2 so I'm currently looking for a way around it.

btw, I didn't make a draft PR yet since I didn't want to spam too much, but perhaps it would be good to do it so you can watch my process?

whoenig, 2022-08-09:
Either way is fine. If you need help/suggestions a draft PR would certainly be helpful to keep the discussion concrete:-)

knmcguire, 2022-08-09:
Alright, I'll work on making one today. I should be close to finishing, as I got some insights from discussion with kristoffer today

knmcguire, 2022-08-28:
So this has been completed, and I also went the extra mile for the dynamic logblock adding and removal, and dynamic parameter  changing, which wasn't the plan for the beginning I see now 😄 oh well

Anyway, there are some things to still fix up but I'll make different issues on that since at least the python server is working so far.

closed 2022-08-28 by knmcguire