IMRCLab/crazyswarm2 #45: /cmd_vel for hover

knmcguire, 2022-07-20:
Hi!

Just something that I would like to ask. I noticed that the twist messages in the topic /cmd_vel are actually used for [roll pitch yaw thrust](https://github.com/IMRCLab/crazyswarm2/blob/8b7444b1858bd26efdb6a6c5f4b6893e8e54341a/crazyswarm2/src/crazyswarm2_server.cpp#L1036-L1049) control, while in other platforms this is used for body fixed velocity control.

Wouldn't it be more logical to use this topic for [the send hover commands](https://github.com/IMRCLab/crazyswarm2/blob/8b7444b1858bd26efdb6a6c5f4b6893e8e54341a/crazyswarm2/src/crazyswarm2_server.cpp#L1099-L1107) instead? Than it will be consistent with other ROS packages

whoenig, 2022-07-20:
I think the best would be to only define custom messages that are clearly
documented rather than reusing twist messages:-) I think we did so for the
other cmd. Since we have our own teleop node, it shouldn't be a
compatibility problem.

knmcguire, 2022-07-20:
true, but perhaps we could consider also having at least a twist based /cmd_vel as well as that will make it easier to connect it to other packages outside of the crazyswarm package.

How about we make a custom message for the RPYT commands that listens to other topic like /cmd_rpty that is more for internal crazyswarm use, and we will have another function that send hover commands based on twist messages on /cmd_vel from external packages

knmcguire, 2022-07-20:
another option, perhaps not something I  am the biggest fan of but perhaps a way to really seperate it from the server, that we make a crazyflie bridge node that turns /cmd_vel commands into /hover_cmd topics, which people can put in the launchfile in case they want to use these external packages.

whoenig, 2022-08-03:
I think we can keep the functionality in the server, but perhaps name the topic something that avoids users using it, if they have the choice, e.g., `cmd_vel_legacy`.

closed 2022-10-04 by knmcguire