IMRCLab/crazyswarm2 #46: Parameter handling for the py server

knmcguire, 2022-07-27:
hi!

I've been trying to figure out how to overwrite the parameters in rclpy based on the yaml file created in [this launch file](https://github.com/IMRCLab/crazyswarm2/blob/8b7444b1858bd26efdb6a6c5f4b6893e8e54341a/crazyswarm2/launch/launch.py#L56-L58) 

It seems that this is being handled in cpp by this line in the crazyflie cpp server:
https://github.com/IMRCLab/crazyswarm2/blob/8b7444b1858bd26efdb6a6c5f4b6893e8e54341a/crazyswarm2/src/crazyswarm2_server.cpp#L478-L481

Unfortunately, the same functions seemed not to be opened up for rclpy, at least to my knowledge. 

What would be a good thing to handle this?
1- Make an cpp node that makes this parameter file into global variables
2- Read out the yaml files directly in the python node and retrieve the names and set the parameters in there. 
3- Someway to get this to properly work in python (perhaps I haven't looked carefully or used the wrong search terms for it)

whoenig, 2022-07-27:
From what I remember, rclpy has some strange limitations when it comes to parameters. One feature that doesn't exist yet, is the callback when parameters change if I remember correctly.
For this particular case, I would go with option 2.

knmcguire, 2022-07-27:
ah yes, that is indeed what I tried and what I feel is the cleanest until there will be more functionality for the python node.  Nothing we can do if the functionality does not exist yet! Thanks for confirming.

Now I get the crazyflie uris like this in the new node:
```
        crazyflies_yaml = os.path.join(
        get_package_share_directory('crazyswarm2'),
        'config',
        'crazyflies.yaml')

        with open(crazyflies_yaml) as f:
            data = yaml.safe_load(f)
        self.uris = []
        cf_dict = {}
        for crazyflie in data:
            uri = data[crazyflie]['uri']
            self.uris.append(uri)
```

closed 2022-07-27 by knmcguire