IMRCLab/crazyswarm2 #50: Additional installation instructions or updates

knmcguire, 2022-08-05:
This is just a list for now of things that are still missing upon a clean install:

- [ ] WSL instructions, since I've been dominantly using that myself for ROS2 now
- [ ] Udev instructions for the crazyradio https://www.bitcraze.io/documentation/repository/crazyflie-lib-python/master/installation/usb_permissions/
- [ ] I didn't need to do it myself but probably these instructions need to be included too? https://crazyswarm.readthedocs.io/en/latest/configuration.html#update-firmware

I'll update this list once I come across more so you can keep it open for now.

whoenig, 2022-08-05:
Thanks! For Crayzswarm(1) we were never able to support WSL, because it lacked USB support for the Crazyradio. Has this changed in WSL in the meantime?

knmcguire, 2022-08-05:
There is now this program called usbipd-win which makes it possible to connect a usb directly go wsl. Works pretty good!

commit e8808b98 (2023-11-16): add usb permission and update instructions

Should fix #50
closed 2023-11-16 by knmcguire