IMRCLab/crazyswarm2 #53: Naming for topics/services that affect all Crazyflies

whoenig, 2022-08-09:
Currently, we use "<cfname>/takeoff" to takeoff a single specific crazyflie and "/takeoff" to takeoff all robots. The naming is a bit confusing. Ideas:

1. Use "/all/takeoff". Pros: short, easy to understand. Cons: Seems to not integrate well with other packages (pollutes the global namespace)
2. Use "/all_cfs/takeoff". Pros: still short and easy to understand. Cons: Still some pollution.
3. Move everything to be part of "crazyflie_server", e.g., "/crazyflie_server/<cfname>/takeoff" and "/crazyflie_server/all/takeoff". Pros: No name pollution. Clear indication which package offers this service. Cons: Very verbose. For many users the existence of the "server" might not even be relevant.

Any thoughts @knmcguire ?

knmcguire, 2022-08-09:
hmm good question... and a difficult one.

for 2, with the currently merged #51, shouldn't it be then all_robots ?

I do feel, even though I agree its a bit convoluted, that option 3 is more ROS2 convention. Also we could use a tilde to have it done automatically, and users who don't like it, could use remapping.  https://design.ros2.org/articles/topic_and_service_names.html

so then it would look like: /crazyflie_server/all_robots/takeoff

knmcguire, 2022-08-09:
after some dinner and some more thought about it, I retract my previous statement 😅 

 I actually think number 2) is better. And then if somebody want to combine it with anything else, they can make their own launch file with a namespace to it.

whoenig, 2022-08-10:
If our recommendation for integration with other robots are namespaces, then perhaps option 1 is the nicest after all?

I would just need to try if it indeed works in ROS2 - I only tried namespaces in ROS1...

knmcguire, 2022-08-10:
hmm yeah that is right. As I told before, I like the ring of /all to indicate all, but I voted 2  to make it consistent with the paramater structure changes of recent

closed 2022-08-11 by knmcguire