IMRCLab/crazyswarm2 #57: dots or slashes for parameters

knmcguire, 2022-08-11:
In PR #52 we had a discussion about dots or slashes for parameters so here a ticket for that. The question was what the convention is in ROS2 for parameters as we get mixed messages from these [docs](https://docs.ros2.org/latest/api/rclcpp/classrclcpp_1_1Node.html#a5ebda51ed6fa4ce8f2cddc9f77e9674e) and [prs](https://github.com/ros2/geometry2/issues/286)

- dots like in namespace.param.subparam
- slashes like in namespace/param/subparam

I've asked a question on https://answers.ros.org/question/404895/parameters-ros2-convention-dots-or-slashes/. The first person seems to think those are dots too but the links he is providing are not super clear either:
- [Design docs for ros parameters](https://design.ros2.org/articles/ros_parameters.html) suggest slashes but is last updated in 2019
- [API doc for roscpp](https://docs.ros2.org/latest/api/rclcpp/classrclcpp_1_1Node.html#a5ebda51ed6fa4ce8f2cddc9f77e9674e) suggest dots and is newer.

Anyway, let's hope somebody from the ROS design team answers soon.

knmcguire, 2022-09-18:
It has been a bit dormant for a while... and it seems there is no clear consensus on it either. I'll try to investigate a bit at ROScon but the nice feature of the current application, is that we can distinquish the 'startup' parameters and the dynamic parameters with dots and slashes so that is perhaps handy?

whoenig, 2022-12-17:
Did you discover anything at ROSCon @knmcguire ? I noticed that when using automatically_declare_parameters_from_overrides (flag in the Node class), ROS2 seems to automatically assume dots and not slashes. Given that it might be useful to use this flag so that other nodes have access to the crazyflies.yaml (I needed this in 71d4a622e238470d8c6c08a0d850f7f814dbbdbc, for example) and for consistency it seems like dots make more sense?

knmcguire, 2022-12-19:
No... unfortunately the verdict is still out on this and i have not found any consistency with other packages. The question did get some answers but the same thing applies: https://answers.ros.org/question/404895/parameters-ros2-convention-dots-or-slashes/

I don't know if we want to keep distinguishing between parameters set at runtime or startup, but since it is up to us as maintainers of this package.... let's say that we choose dots!

closed 2022-12-19 by whoenig