IMRCLab/crazyswarm2 #79: decouple motion capture tracking package

knmcguire, 2022-10-03:
Currently when crazyswarm2 has to be installed from scratch, the motion capture tracking package is mandatory due to this:

```

CMake Error at CMakeLists.txt:17 (find_package):
...
  Could not find a package configuration file provided by
  "motion_capture_tracking_interfaces" with any of the following names:

    motion_capture_tracking_interfacesConfig.cmake
    motion_capture_tracking_interfaces-config.cmake
```

Shouldn't this be an option based on what people are using as a positioning system? Perhaps also in the combination of what I mentioned in #67 ?

Not an urgency now but a point of discussion in the future :)

knmcguire, 2022-10-06:
another reason to decouple it.. I am now trying out crazyswarm2 on 22.04 and humble and getting the following issue of motion capture:

```
CMake Error at deps/libmotioncapture/deps/pybind11/tools/pybind11NewTools.cmake:196 (python3_add_library):
  Unknown CMake command "python3_add_library".
Call Stack (most recent call first):
  deps/libmotioncapture/CMakeLists.txt:158 (pybind11_add_module)

```

whoenig, 2024-01-30:
We are always needing the motion_capture_interfaces package, because the server needs to include and link against it. Given that this is now released in humble/iron/rolling, I think this will be less annoying and I vote for closing this issue.

knmcguire, 2024-01-31:
absolutely. So we could change the installation instructions now I assume? https://imrclab.github.io/crazyswarm2/installation.html

whoenig, 2024-01-31:
I think we would need wait for the next humble sync. It's in iron and rolling, but didn't make the cut for humble binary release, yet.

knmcguire, 2024-02-01:
ah gotcha. I'm closing the issue but let's not forget to update the documentation once it has been synced

closed 2024-02-01 by knmcguire