IMRCLab/crazyswarm2 #87: Fix symlink issue in crazyflie_examples

knmcguire, 2022-10-06:
Currently, if anybody changes anything in crazyflie_examples's launch file, config files or data files, they will need to reinstall with colcon build which is annoying.

Currently I've added a warning to the text in #86, but there are other things we can do:

- Convert the entire package to cmake. I tried to do this with python files only but was unsuccessful
- Hope that https://github.com/colcon/colcon-core/issues/407 gets resolved at one point

Anyway, this issue will also warn crazyswarm2 users that this is still a problem.

whoenig, 2022-10-06:
What do you mean by "unsuccessful" - did it not build, or did the problem remain? I am pretty sure I have used the cmake-trick in other projects successfully.

knmcguire, 2022-11-16:
Yeah what I meant with unsuccessful is that it didn't build for me, but I probably forgot something in the pipeline of implementing it. 

I'll review the current PR!

commit fd778f30 (2022-11-17): Issue87 (#111)

* examples: switch to cmake for symlink-install

Workaround for https://github.com/colcon/colcon-core/issues/407

Fixes #87
closed 2022-11-17 by whoenig