IMRCLab/crazyswarm2 #90: Installation documentation

ywang760, 2022-10-19:
I'm new to crazyswarm2 and went through the crazyswarm installation documentation. 
On the fourth step in first installation, py_crazyswarm2 can't be found. I suspect it's crazyflie_py.

knmcguire, 2022-11-02:
Hi! Sorry for the late reply.

That is strange, I can not find a build for py_crazyswarm2 in any of the cmakefiles in the latest main branch. Are you sure you have updated the repo and deleted the build and install folders before `colcon build` ?

knmcguire, 2022-11-09:
I'll close this as I can not reproduce it and inactivity, but please reopen if it is still a problem. Also for more generic help, use https://github.com/IMRCLab/crazyswarm2/discussions

closed 2022-11-09 by knmcguire