IMRCLab/crazyswarm2 #103: update crazyflie firmware for updated python bindings

knmcguire, 2022-11-09:
In order for me to start implementing the webots simulation node, we will need to have the latest updates of the crazyflie-firmware and python bindings. Unfortunately, updating it now causes this error when colon building it:

```
Starting >>> cffirmware
--- stderr: cffirmware                                                   
/home/kimberly/Development/collaborations/crazyswarm2/ros2_ws/src/crazyswarm2/crazyflie-firmware/bindings/setup.py:3: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  import distutils.command.build
cc1: fatal error: build/cffirmware_wrap.c: No such file or directory
compilation terminated.
```

knmcguire, 2022-11-09:
@whoenig, I guess that it needs to be configured to the latest kbuild right? I just can't really find the makefile responsible for this yet but I haven't looked very hard yet

whoenig, 2022-11-09:
I think the problem is that colcon shouldn't do anything with this folder. You could use
```
colcon build --symlink-install --packages-ignore=crazyflie-firmware
```

Ultimately, we probably don't want the firmware as a submodule at all, but as a pip requirement instead.

knmcguire, 2022-11-09:
I agree.. it shouldn't be a part of it in the first place. But until it becomes available as a pip package, what do we want to do with it in the mean time? It would be nice to have the simulator available rather sooner then later.

Maybe for the time being, we can precompile the python bindings and put them in the simulation package?

knmcguire, 2022-11-09:
ah sorry I misunderstood you, yes that will work for now

closed 2022-11-25 by whoenig