IMRCLab/crazyswarm2 #117: Simulation Support

whoenig, 2022-11-17:
As part of Crazyswarm2, we want a new improved simulation support.

# Goals

We would like the following features

- Operates as a backend, i.e., can be used with C++ and Python nodes (unlike Crazyswarm1, which only supported the high-level Python API)
- Supports multiple backends, including ones with a physics engine (unlike Crazyswarm1, which only supported visualization)
- Uses the Crazyflie firmware as software-in-the loop (SIL) (similar to Crazyswarm1)
- Can be used for heterogeneous simulation, e.g., UAVs and ground robots (with some backends; unlike Crazyswarm1, which did not have such support)

As simulation backends we are considering:

1. A pure visualization (e.g., Rviz, meshcat, or matplotlib)
2. WeBots, see also #106 and #77
3. Our custom simulator (currently at https://github.com/IMRCLab/pyCrazyflie)
4. Gazebo
5. CoppeliaSim

# Usage (user-perspective)

From a user-perspective, the simulator should be usable using (or similar)

```
ros2 launch crazyflie launch.py backend:=sim sim:=webots
```

which will either connect or launch to the existing simulator, spawn the Crazyflies at the specified initial positions in the launch file, and react to ROS topics/services as similar as possible as when using the physical backend.

# Design

- The simulator will be written in Python and uses the crazyflie-firmware Python bindings to support SIL.
- The interface of the simulator is the same as with the other backends, i.e., it provides high-level services (such as takeoff) for individual drones and all drones, as well as low-level topics (such as cmd_vel).
- To support multiple simulator backends, the simulator will just map between the ROS interface and desired states (for visualization) and motor forces (for physics-based simulation). Each simulation backend will need to be able to provide:
    - a function to initialize a UAV at a given state/position. Input: state, Output: UAV-ID
    - a stepUAV RPC (can be a Python call, ROS service, or another RPC approach). Input: UAV-ID, desired states, motor forces, Output: new state (or some error if the drone crashed). Note that the new state is not strictly needed, but might allow us to have standardized logging tools across backends.
    - a stepTime RPC, to advance simulation time. Input: dt

# Open Issues

We need to understand the way simulation time works in ROS2. One resource is https://design.ros2.org/articles/clock_and_time.html.

whoenig, 2022-11-17:
Adding @knmcguire and @Khaledwahba1994 to comment on this proposal.

whoenig, 2022-11-25:
The basic infrastructure with visualization backend has been merged. The next step is to enable controllers from the Python bindings to support physics-based backends.

knmcguire, 2025-04-25:
This is already available.

closed 2025-04-25 by knmcguire