IMRCLab/crazyswarm2 #130: cflib backend: odom logging improvements

whoenig, 2022-11-25:
The current odom topic is defined here: https://github.com/IMRCLab/crazyswarm2/blob/main/crazyflie/scripts/crazyflie_server.py#L484-L524

There seem to be two issues with it:

- This seems to be a 2D version only, which is counterintuitive. I can see some value, but perhaps we need to rename this to odom2d, and we can think of having a real odometry message for the full state (which is similar in spirit as /pose)
- Instead of "kalman.PX", one should use "stateEstimate.vx" etc. (to be independent of the filter used)

@knmcguire for comment:-)

knmcguire, 2022-11-25:
The 2d odom message is in order to be compatible with the 2d SLAm toolbox, where at least it is common to have the odometry of the footprint aka shadow of the odometry. Isn't it enough for now to say that pose is meant for 3d and odometry is meant for 2d? Also i would say that adding a Z axis might not even really matter for intergration actually

[Kalman.px ](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/api/logs/#kalmanstatepx) is in body fixed coordinates while stateEstimate.vx is in world fixed coordinates, so I wanted to avoid the body rotation everytime I get the msg, and since the EKF is the only full state filter there... perhaps it is okay for now?

whoenig, 2022-11-25:
Pose only has the first-order, odometry also the second order information (i.e., velocity). I think there might be a use case for /odom in 3d in the future. Perhaps we can rename to /odom2d, and then use re-mapping to use the toolbox?

Good point with Kalman.px - thanks for the clarification!

knmcguire, 2022-11-28:
Ah yes that (odometry being pose and velocity( I agree with, but it seems that at least the toolboxes I connected it with only used the position part of it. 

I think that probably would be a good option for remapping indeed, or call it /odom_footprint or something. On the other hand... we could also just add the velocity height and height to the odom message and see if that would cause any issues with the SLAM toolbox and Navigation tutorials, because if there aren't any issues we can just do full 3d odom message and make it easier for everybody. 

I'll try to find some time this week to work on it.

whoenig, 2022-12-05:
Yes - That's why the compressed state was added a while ago. The only downside is that the range is more limited, but for most set ups this should suffice.

whoenig, 2022-12-07:
Very interesting idea! This hasn't been implemented on the C++ side, but might be worthwhile doing. Is there any documentation that explains the precision difference between float32 and FP16? I.e., would that be even a more reasonable alternative to stateEstimateZ (and similar double variables, like pm.vbatmv vs. pm.vbat)?

knmcguire, 2022-12-07:
Actually I got it from the logging example in the crazyflie python library: https://github.com/bitcraze/crazyflie-lib-python/blob/1c07945e7f93587d9e5afd860258aa304b61555d/examples/logging/basiclog.py#L82 

here it was used on pm.vbat indeed.

There not really that much more documentation... but I found this in the firmware:
https://github.com/bitcraze/crazyflie-firmware/blob/4f481aba69f577f29030d03668cfabd9c2b852b2/src/utils/src/num.c#L32-L46

knmcguire, 2022-12-19:
I guess we can close this now right now that #146 has been merged?

whoenig, 2022-12-19:
Agreed:-)

closed 2022-12-19 by whoenig