IMRCLab/crazyswarm2 #140: Installation Issues.

santiagorg2401, 2022-12-03:
Hi, I hope you're doing alright,
I'm recently installing for the first time as a former Crazyswarm1 developer, I've faced an issue regarding the crazyflie-firmware Python bindings whenever I try to compile it.
As I run `make bindings_python` I get:

```
Makefile:219: warning: overriding recipe for target 'bindings_python'
tools/kbuild/Makefile.kbuild:147: warning: ignoring old recipe for target 'bindings_python'
make[1]: Entering directory '/home/santiagorg2401/crazyflie-firmware/build'
  GEN     ./Makefile
scripts/kconfig/conf  --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
make[3]: *** [../scripts/kconfig/Makefile:38: silentoldconfig] Error 1
make[2]: *** [tools/kbuild/Makefile.kbuild:403: silentoldconfig] Error 2
make[1]: Nothing to be done for 'bindings_python'.
make[1]: Leaving directory '/home/santiagorg2401/crazyflie-firmware/build'
swig -python -Isrc/modules/interface -Isrc/hal/interface -Isrc/utils/interface -o build/cffirmware_wrap.c bindings/cffirmware.i
python3 bindings/setup.py build_ext --inplace
/home/santiagorg2401/crazyflie-firmware/bindings/setup.py:3: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  import distutils.command.build
running build_ext
building '_cffirmware' extension
creating build/temp.linux-x86_64-3.10
creating build/temp.linux-x86_64-3.10/build
creating build/temp.linux-x86_64-3.10/src
creating build/temp.linux-x86_64-3.10/src/modules
creating build/temp.linux-x86_64-3.10/src/modules/src
creating build/temp.linux-x86_64-3.10/src/utils
creating build/temp.linux-x86_64-3.10/src/utils/src
x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Isrc/modules/interface -Isrc/hal/interface -Isrc/utils/interface/lighthouse -Isrc/utils/interface -Ibuild/include/generated -Isrc/config -Isrc/drivers/interface -Isrc/platform/interface -I/usr/include/python3.10 -c build/cffirmware_wrap.c -o build/temp.linux-x86_64-3.10/build/cffirmware_wrap.o -O3 -fno-strict-aliasing -Wno-address-of-packed-member
x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Isrc/modules/interface -Isrc/hal/interface -Isrc/utils/interface/lighthouse -Isrc/utils/interface -Ibuild/include/generated -Isrc/config -Isrc/drivers/interface -Isrc/platform/interface -I/usr/include/python3.10 -c src/modules/src/attitude_pid_controller.c -o build/temp.linux-x86_64-3.10/src/modules/src/attitude_pid_controller.o -O3 -fno-strict-aliasing -Wno-address-of-packed-member
In file included from src/modules/src/attitude_pid_controller.c:35:
src/platform/interface/platform_defaults.h:30:10: fatal error: autoconf.h: No such file or directory
   30 | #include "autoconf.h"
      |          ^~~~~~~~~~~~
compilation terminated.
error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
make: *** [Makefile:220: bindings_python] Error 1
```

I've already tried cloning the [crazyflie-firmware](https://github.com/bitcraze/crazyflie-firmware) repo as said in their documentation with the `--recursive` flag without success:
`$ git clone --recursive https://github.com/bitcraze/crazyflie-firmware.git`

I also followed the installation instructions in https://imrclab.github.io/crazyswarm2/installation.html and it did not work.

```
git clone https://github.com/bitcraze/crazyflie-firmware.git
make bindings_python
export PYTHONPATH=<replace-with-path-to>/crazyflie-firmware:$PYTHONPATH
```

My system information is:
* Ubuntu 22.04.
* ROS Humble.

Thank you in advance.

santiagorg2401, 2022-12-03:
Update:
It worked using:
```
sudo apt-get install make gcc-arm-none-eabi
git clone --recursive https://github.com/bitcraze/crazyflie-firmware.git

cd crazyflie-firmware/
make defconfig
make bindings_python
```

I think it should be noted on the docs unless this is something that is assumed to be known.

knmcguire, 2022-12-03:
ah yes good catch!

Maybe it would be better to instead add a piece of doc of installing the python bindings in the crazyflie-firmware documentation instead and refer directly to that. I'm always a bit careful when it comes to double doc and I think it needs to be as close to the code as possible. What do think @whoenig?

whoenig, 2022-12-03:
I agree that having a direct link to the official documentation would be better. However, if that takes to long, we can also fix it here in the meantime and hope that the pip-bindings are coming soon:-)

knmcguire, 2022-12-05:
Agreed. I'll have a documentation rush on the bitcraze website soon so I'll take this with that, but if it is not fixed by the end of the week lets add it to the crazyswarm2 doc

knmcguire, 2022-12-15:
It should be part of these instructions now! https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/building-and-flashing/build/#build-python-bindings. I'll add them to the instructions

PR #155 (2022-12-15): update python binding installation instructions
fixes #140
closed 2022-12-15 by knmcguire