IMRCLab/crazyswarm2 #149: Swarm Management / Monitoring

whoenig, 2022-12-13:
In Crazyswarm1, the swarm management was done using `chooser.py`, which allows enabling/disabling robots, reboot, battery check, and firmware updates. Conceptually, chooser.py is a tool that reads and write config files and calls other CLI programs to report the result in a GUI. The major drawback of such an approach is that it can only work if crazyflie_server itself is not running. This makes the workflow quite unintuitive, because even disabling a single drone requires shutting down the server, potentially rebooting the drones, and then reconnecting again.

In Crazyswarm2, this workflow should be simplified, by adding the necessary logic directly to the server. Specifically:

* A new service, that allows to reboot a Crazyflie
* A service to enable/disable Crazyflies dynamically (note that we do not necessarily need a way for a full dynamic addition, i.e., we can assume that Crazyflies are declared in crazyflies.yaml)
* A default topic for typical monitoring tasks, such as battery voltage, link quality, and ability to fly (sensors calibrated? emergency on?)
* An updated GUI tool, that operates like chooser.py, but can be used (only) if the server is running by using the services and topics mentioned above. Note that firmware updated would work by: a) disabling the Crazyflie, b) running cfloader with the appropriate commands, and c) enabling the Crazyflie again. This avoids that any backend needs to have specialized knowledge on how to execute a firmware update.

whoenig, 2022-12-13:
Adding @knmcguire, to comment on the above design suggestion.

knmcguire, 2025-04-25:
already added with the webgui

closed 2025-04-25 by knmcguire