IMRCLab/crazyswarm2 #167: sim - np backend - multi robot support

whoenig, 2023-01-04:
Currently, we only properly support single UAVs in simulation with physics backend due to firmware Python bindings limitations. After the firmware is fixed, this needs some more testing

matthewoots, 2023-01-09:
Hi @whoenig, can I know the progress for this or where can I help if possible? So the issue is in `crazyflie-firmware`?

whoenig, 2023-01-09:
For the Mellinger controller, I already created a PR for the firmware at https://github.com/bitcraze/crazyflie-firmware/pull/1180. Once that is merged, there will be tiny changes here and some testing required. I don't think it's easily possible to help yet. Once the mellinger controller is merged, we would need similar firmware refactoring for any other controller that we want to support properly.

matthewoots, 2023-01-15:
Hi @whoenig thank you for the pull request to create a mellinger controller obj. I think https://github.com/IMRCLab/crazyswarm2/pull/171 will solve this issue. Thanks!

whoenig, 2023-01-16:
Agreed - thanks!

closed 2023-01-16 by whoenig