IMRCLab/crazyswarm2 #173: server cpp: not all parameters from crazyflies.yaml might be set

whoenig, 2023-01-21:
When connecting to multiple Crazyflies, not all parameters that are specified in crazyflies.yaml are actually updated on all drones. This behavior is not deterministic and it seems like the first Crazyflie with connection attempts is more likely to have issues. It seems to be related to the ROS2 parameter callback which is not always called in that case.

To reproduce: connect to at least three drones and set about 30 params in crazyflies.yaml

knmcguire, 2023-01-23:
A follow up question! Does this happen only with the cpp backend or also with the python backend?

whoenig, 2023-01-23:
I only tested with the cpp backend

knmcguire, 2023-01-23:
ah oke good to know! I have tested this with multiple crazyflies with the cflib backend but never with 30 parameters. I think it is still okay but always good to check

knmcguire, 2023-01-24:
I am unable to create this myself with both the cpp and the cflib backend, but the crazyradio was literally next to to crazyflies and I've only tried it about 3 times. But I think the solution you proposed is quite good anyway to set it two one crazyflie at the time.

PR #177 (2023-01-25): server cpp: refactor initial parameter update
* Use direct function calls for the initial set up, rather than relying on ROS2 events
* Fixes #173
closed 2023-01-25 by whoenig