IMRCLab/crazyswarm2 #184: Hello world example not working!

Abhinandan-Krishnan, 2023-02-08:
I am trying to run hello world example. I can see the crazyflie on rviz and see the poses topic as well. When I run the example, the crazyflie fails to reach the targeted altitude and bounces at a small altitude and stops. I can see the similar behavior in  RVIZ as well. The battery is fully charged.

knmcguire, 2023-02-09:
Hi!

We need more information about your problem

- Have you set anything different in the crazyflie.yaml in terms of parameters than the standard file (perhaps share it)
- Does this happen only with the cpp backend or the cflib backend too?
- Which version of the Crazyflie firmware are you using
- Which positioning system are you using?

Abhinandan-Krishnan, 2023-02-14:
Thank you for your help!

1) I haven’t changed any parameters in the crazyflie.yaml file.
I am attaching a doc version as git doesn’t allow me to attach a .yaml file.
[](url)
[crazyflie_parameters_yaml_copy.docx](https://github.com/IMRCLab/crazyswarm2/files/10711921/crazyflie_parameters_yaml_copy.docx)

2) Similar problem with cpp backend as well. The quad is bouncing. I can see the difference in motor speed during takeoff, hover and landing actions. 

3) I am attaching the screenshot from cfclient Console tab which has the firmware version. updated the crazyradio firmware as well to the latest version.
[](url)
[image]

4) Using Vicon Motion Capture system. Same case for libRigidBodyTracker, motionCapture

knmcguire, 2023-02-14:
Hi!

I think I see what is going on already.  The yaml file is changed compared to the [default yaml file](https://github.com/IMRCLab/crazyswarm2/blob/main/crazyflie/config/crazyflies.yaml), so his part is important
```
  firmware_params:
    commander:
      enHighLevel: 1
    stabilizer:
      estimator: 0 # 0: Auto select, 1: complementary, 2: kalman
      controller: 0 # Auto select(0), PID(1), Mellinger(2), INDI(3), Brescianini(4) 
```

If you select `estimator: 0`, that meanst that the estimator is based on which positioning deck is being attached. Since you are using Mocap, there is no 'special deck' attached so the estimator is using the default complementary filter which can not handle position state. The kalman can handle position state.

In your case, please set estimator back to 2:
https://github.com/IMRCLab/crazyswarm2/blob/f65511ecf2161e1e03a7fc3ff97c0b5b811fbe51/crazyflie/config/crazyflies.yaml#L93

knmcguire, 2023-02-19:
So, did this work for you?

Abhinandan-Krishnan, 2023-02-24:
Sorry for the delayed response. Yes, that worked, thanks for the help!

Also had a follow up question, 
In the launch.py file in crazyflie package, if I am using vicon as motion tracker with unique markers, should I keep this below snippet in the launch file? 

```
motion_capture_params = motion_capture["/motion_capture_tracking"]["ros__parameters"]
motion_capture_params["rigid_bodies"] = dict()

for key, value in crazyflies["robots"].items():
    robot_type = crazyflies["robot_types"][value["type"]]
    if value["enabled"] and robot_type["motion_capture"]["enabled"]:
        motion_capture_params["rigid_bodies"][key] = {
            "initial_position": value["initial_position"],
            "marker": robot_type["motion_capture"]["marker"],
            "dynamics": robot_type["motion_capture"]["dynamics"]
        }
```

knmcguire, 2023-02-24:
Hi! Please start a new discussion since this issue is now resolved. Also your questions should be at the discussions instead: https://github.com/IMRCLab/crazyswarm2/discussions. That is the main spot for support questions

closed 2023-02-24 by knmcguire