IMRCLab/crazyswarm2 #188: How to determine raw marker positions for marker configuration

zjwilliams20, 2023-02-15:
In `crazyswarm`, I remember using a `mocap_helper` node that enabled me to determine the marker configurations for our drones. I don't see this implemented in `crazyswarm2` nor in the documentation.
Do y'all have any recommendations on how to set this? I tried going through the ros topics after launching like `/pointCloud` and `/poses`, but these don't provide the raw marker configurations in the world frame that I need.

whoenig, 2023-02-15:
Technically `/pointCloud` has all you need, but at least `ros2 topic echo` doesn't decode the format. I usually help myself by running the example of `https://github.com/IMRCLab/libmotioncapture/tree/main` directly, which prints out everything you need.

A proper way of handling this is tracked in https://github.com/IMRCLab/motion_capture_tracking/issues/2, so I think we can close the issue in this repo.

closed 2023-02-15 by whoenig

zjwilliams20, 2023-02-17:
Roger. That seemed to work well whenever our Vicon system feels like cooperating. Thanks for the quick response!