IMRCLab/crazyswarm2 #225: cmdFullState in crazyflie_sil.py

Khaledwahba1994, 2023-03-22:
It seems that the cmdFullState is not called whenever it is used for simulation along with any of the example scripts.
The output for running: 
```bash
ros2 run crazyflie_examples cmd_full_state --ros-args -p use_sim_time:=True
```
is
```bash
[crazyflie_server-3]   File "/home/khaledwahba94/imrc/ros2_ws/install/crazyflie_sim/lib/crazyflie_sim/crazyflie_server", line 33, in <module>
[crazyflie_server-3]     sys.exit(load_entry_point('crazyflie-sim==2.0.0', 'console_scripts', 'crazyflie_server')())
[crazyflie_server-3]   File "/home/khaledwahba94/imrc/ros2_ws/install/crazyflie_sim/local/lib/python3.10/dist-packages/crazyflie_sim/crazyflie_server.py", line 318, in main
[crazyflie_server-3]     rclpy.spin(crazyflie_server)
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
[crazyflie_server-3]     executor.spin_once()
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 713, in spin_once
[crazyflie_server-3]     raise handler.exception()
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[crazyflie_server-3]     self._handler.send(None)
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
[crazyflie_server-3]     await call_coroutine(entity, arg)
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 343, in _execute_subscription
[crazyflie_server-3]     await await_or_execute(sub.callback, msg)
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[crazyflie_server-3]     return callback(*args)
[crazyflie_server-3]   File "/home/khaledwahba94/imrc/ros2_ws/install/crazyflie_sim/local/lib/python3.10/dist-packages/crazyflie_sim/crazyflie_server.py", line 303, in _cmd_full_state_changed
[crazyflie_server-3]     self.cfs[name].cmdFullState(
[crazyflie_server-3]   File "/home/khaledwahba94/imrc/ros2_ws/install/crazyflie_sim/local/lib/python3.10/dist-packages/crazyflie_sim/crazyflie_sil.py", line 168, in cmdFullState
[crazyflie_server-3]     self.setState.pos = firm.mkvec(*pos)
[crazyflie_server-3] AttributeError: 'method' object has no attribute 'pos'
[ERROR] [crazyflie_server-3]: process has died [pid 301334, exit code 1, cmd '/home/khaledwahba94/imrc/ros2_ws/install/crazyflie_sim/lib/crazyflie_sim/crazyflie_server --ros-args -r __node:=crazyflie_server --params-file /tmp/launch_params_bzywojq2 --params-file /tmp/launch_params_9ul_0dv2'].
```

mrfikri-ai, 2023-08-16:
Hi, I got the same issue when running cmd_full_state example scripts.
Do you have any more information to overcome this problem?

mwsatman, 2023-08-23:
Hi I got the same issue as this one 

`[crazyflie_server-3] Traceback (most recent call last):
[crazyflie_server-3]   File "/home/localadmin/ros2_ws/install/crazyflie_sim/lib/crazyflie_sim/crazyflie_server", line 33, in <module>
[crazyflie_server-3]     sys.exit(load_entry_point('crazyflie-sim==2.0.0', 'console_scripts', 'crazyflie_server')())
[crazyflie_server-3]   File "/home/localadmin/ros2_ws/install/crazyflie_sim/local/lib/python3.10/dist-packages/crazyflie_sim/crazyflie_server.py", line 318, in main
[crazyflie_server-3]     rclpy.spin(crazyflie_server)
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
[crazyflie_server-3]     executor.spin_once()
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 713, in spin_once
[crazyflie_server-3]     raise handler.exception()
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[crazyflie_server-3]     self._handler.send(None)
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
[crazyflie_server-3]     await call_coroutine(entity, arg)
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 343, in _execute_subscription
[crazyflie_server-3]     await await_or_execute(sub.callback, msg)
[crazyflie_server-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[crazyflie_server-3]     return callback(*args)
[crazyflie_server-3]   File "/home/localadmin/ros2_ws/install/crazyflie_sim/local/lib/python3.10/dist-packages/crazyflie_sim/crazyflie_server.py", line 303, in _cmd_full_state_changed
[crazyflie_server-3]     self.cfs[name].cmdFullState(
[crazyflie_server-3]   File "/home/localadmin/ros2_ws/install/crazyflie_sim/local/lib/python3.10/dist-packages/crazyflie_sim/crazyflie_sil.py", line 168, in cmdFullState
[crazyflie_server-3]     self.setState.pos = firm.mkvec(*pos)
[crazyflie_server-3] AttributeError: 'method' object has no attribute 'pos'
[ERROR] [crazyflie_server-3]: process has died [pid 24152, exit code 1, cmd '/home/localadmin/ros2_ws/install/crazyflie_sim/lib/crazyflie_sim/crazyflie_server --ros-args -r __node:=crazyflie_server --params-file /tmp/launch_params_y7fwhh5i --params-file /tmp/launch_params_pksi46tr'].
`
How to deal with this?

commit 3b91a7d5 (2023-08-28): [sim] fix issue when using cmdFullState

Fixes #225
closed 2023-08-28 by whoenig