IMRCLab/crazyswarm2 #235: RVIZ transform mocap pose conflict

knmcguire, 2023-03-28:
I noticed that when I got the pose logging enabled (to check if Mocap data was actually sent through to the Crazyflie), that the transform for Mocap (with rigid bodies) and Pose transform has the same name. MoCaps' rigid body names needs to be same as crazyflie.yaml in order for the motion capture package to interpret that to be the right rigidbody/ marker for the Crazyflie and it will make sense for the transform directly from the server to be the same name as well, but currently in RVIZ you'll see 'cf1' constantly swift between the MoCap contributed transform and the Pose logging contributed transform. 

I would imagine that we would like to distinguish external positioning from the internal estimated positioning or... ? And if we want to change the naming, which transform name should we augment (or both?)

whoenig, 2023-11-16:
One idea:

1. Use cfname in `odom` frame for the on-board data we get via logging
2. Use cfname in `mocap` frame for the transformation we get from the motion capture
3. Have some logic in the launch file that has a static transformation publisher world<->odom and/or world<->mocap (based on the operation mode).

I think the term `mocap` could refer to motion capture, lighthouse, or LPS. However, this might be confusing, in which case we could use `eps` (external positional system) or similar term to indicate that we talk about a common, external coordinate frame.

Any thoughts @knmcguire ? I think you are also more familiar in which cases `odom` is used, so your input would be super valuable here:-)

knmcguire, 2023-11-16:
hi! 

Yeah it is good to seperate them in transforms, but the problem is that usually visually, these transforms are very close to eachother.

That is a good question... we could use 'ext' or 'global', or 'gt' for ground truth. Odom would be used in terms of estimated position usually from non-global position estimates, like wheel odometry or flow, but as soon as the external position measurements come in, the estimated position and the position that is put in will be very much the same.

Let me try to think about this a bit and see how others do it.

whoenig, 2024-01-30:
Did you have any more ideas @knmcguire ? 

Another use-case to consider could be two CFs with flow decks attached and no external positioning system. Both will believe that they are initially at the same spot. Then, a transform to `world_<cfname>` might make sense, or `odom_<cfname>`, since we have to separate that each of them might have their own local world frame and the mapping of the local world frame to the global world frame might or might not be known.

Visualization in rviz will be tricky this way, since we would need static transforms for each world* to world...

whoenig, 2025-05-06:
Thanks for all these great links! I don't quite get "On the crazyswarm2 server side, we grab that same transform, but we substract the state estimate so we get the map->odom transform". Could we perhaps look at what happens (which nodes publishes which transforms) in "common" use-cases:

1. Mocap with 2 robots
   - motion_capture_tracking publishes mocap -> cf1, mocap -> cf2 (not sure why we would publich mocap -> odom here?)
   - server publishes odom_cf1 -> cf1, odom_cf2 -> cf2 (from the onboard EKF)
   - someone needs to publish mocap -> map as identity (?)? perhaps we also need odom -> odom_cf1, odom ->odom_cf2 to be able to see all state estimates in rviz?

2. Lighthouse with 2 robots
   - server publishes odom_cf1 -> cf1, odom_cf2 -> cf2 (from the onboard EKF)
   - perhaps we also need odom -> odom_cf1, odom ->odom_cf2 to be able to see all state estimates in rviz?
 
3. Flow-deck with 2 robots
   - as 2., but we would *not* publish odom -> odom_cf1, odom -> odom_cf2 (unclear how we would decide on that)

4. Hybrid: 1 with mocap, 1 with flow-deck
   - mocap -> cf1, odom_cf1 -> cf1, odom_cf2 -> cf2 (again, no "matching" of odom_cf1 and odom_cf2)

knmcguire, 2025-05-07:
Woops I forgot to link to REP 105: https://ros.org/reps/rep-0105.html

Yes, that is the case, but for mobile robots the wheel odometry can be more separated from the map generation from SLAM. Then you could have the full chain if they use mocap for ground truth to have mocap->map->odom->baselink.  So if the robot generates the map->odom->baselink and the mocap system generates mocap->baselink, then we use that to calculate mocap->map. I said subtracting as in subtracting vectors but it's probably not the right wording 😅 

Also in REP 105, they'd like to see a generated map per robot, also from SLAM, as each of those maps can be different, but it doesn't make it very easy.... 

1- So the idea is that mocap->cf1 will be completely separate transform, but from there we will calculate (same as mocap) map_cf1 -> odom_cf1 and map_cf2->odom_cf2. Mocap -> map_cf would be in this case  zero transforms indeed but not sure how it will look like on the map. But at least that is closer to what's stated in the REP, and there is currently no notion of a 'shared odometry' anyway.
2- For lighthouse it's a bit tough...  I would do map_cf1->odom_cf1 and map_cf2->odom_cf1 but then you won't see the crazyflies at once.
3- Yeah.... this is super tricky... but indeed not publishing odom -> odom_cf1 would be logical the case
4- Yes that seems like it

knmcguire, 2025-05-07:
We might need to step a bit back here... If we do it according to the REPS (the rep147 for aerial is old and deterred btw), we might do it 'right' but we also create a mess on RVIZ2. Since the jury might still be out on how to do this for aerial vehicles / multirobot systems, I would actually argue that for now we make it so it is actually viewable.

I would propose doing it a bit differently for the time being and just keeping it simple, and we can always change later:
- map->cf1 map->cf2 (this is the standard view of the users)
- mocap->cf1 mocap->cf2 (this is what the users can see when they select mocap as main frame on rviz2)

Here map will just mean 'state estimate in the world', since that is basically what map means for mobile robots as well.

knmcguire, 2025-05-07:
Wait... maybe we don't want it to be called `map`, since that will for sure conflict with the SLAM examples and if anyone likes to connect it to nav (they are very picky that the transform odom->cf1 needs to exist). Let's just call it `world` still

knmcguire, 2025-05-12:
Hi! So I've tried a couple of things out and found out 2 things:
* Apparently it is not possible 2 parent-frames to a transform, so my idea of both an mocap->cf and world->cf is not going to work.
* Being fully [REP 105](https://ros.org/reps/rep-0105.html#id8) compliant mostly makes a very messy origin with all kinds of names... I would vote actually against it for now, at least for the first release

So I suggest to do the following:

* Motion capture node: `mocap -> mc_cf1`
* Crazyflie server node: `world -> cf1`

And let's keep thing configurable at least, such that `mocap` can still be changed to another name (like map for the mapping examples) and same goes with `world`

@whoenig what do you think of this? this would require a change on the motioncapture ros2 package as well.

whoenig, 2025-05-13:
So for our former examples we get the following:

1. Mocap with 2 robots
   - motion_capture_tracking publishes mocap -> cf1_mocap, mocap -> cf2_mocap
   - server publishes world -> cf1, world -> cf2 (from the onboard EKF)
   - optionally, publish mocap -> world as identity, so one can see both mocap and onboard estimate in rviz at once

2. Lighthouse with 2 robots
   - server publishes world -> cf1, world -> cf2 (from the onboard EKF)
 
3. Flow-deck with 2 robots
   - this is still unclear. cf1 and cf2 should *not* be both with respect to world, because they don't actually share a coordinate system?

4. Hybrid: 1 with mocap, 1 with flow-deck
   - Also unclear what would be good in this case.

It's not a problem to change the motioncapture package. I think the parent is already configurable. We could also introduce a format-string or prefix or postfix for the actual robot names.

knmcguire, 2025-05-15:
Since it is not really possible to see from the EKF if there is a fixed world positioning, I would propose the following for 3 and 4

3. Technically they should not match, but the user could indicate that as a parameter setting (since the crazyflies don't know this). This will probably be 'experimental' at first but if they do 'relative_localization' to true, that we setup in the server to output:
- odom_cf1 - cf1
- odom_cf2 - cf2
This does mean that the user would need to switch RVIZ3 views to go to odom_cf1, and odom_cf2, or put through a static null transform to do world -> odom_cf1 -> cf1, world -> odom_cf2 -> cf2 which could be done from the launch file, but technically the server would do it correct. 

4. I would still separate them. 1 with mocap as case 1 and the flowdeck one as case 3. Those cases are just incompatible and should be seen as seperate transforms, but the user can always chose to add a static transform to connect mocap->odom_cf2->cf2 to see it in the same rviz2

whoenig, 2025-05-17:
I don't like using a "global" flag like "relative_localization", because we could have mixed systems (e.g., 2 flowdeck, 2 mocap). Perhaps we should then simply add a new key in crazyflies.yaml that has the same "hierarchical" behavior that we have for params and logging? 

Example:
```
robots:
  cf231:
    enabled: true
    uri: radio://0/80/2M/E7E7E7E7E7
    initial_position: [0.0, 0.0, 0.0]
    type: cf21  # see robot_types
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    # firmware_logging:
    #   enabled: true
    #   custom_topics:
    #     topic_name3: 
    #       frequency: 1
    #       vars: ["acc.x", "acc.y"]
    reference_frame: world
```
(and reference_frame can be also in crazyflie_types, or all).

Does that solve all of our problems @knmcguire ?

If so the list of changes would be:

1. Change motion_capture_tracking to allow pre/postfix of frame names; change the config file so that we add a _mocap postfix in crazyswarm2.
2. Add "reference_frame" keys in crazyflies.yaml and support for it in all 3 backends.

knmcguire, 2025-05-19:
oh yes! that would be perfect actually. Good thinking! I'm all on board with this one.

I'll give it a go with the cflib backend and see how that all looks but that sounds like a very good plan