IMRCLab/crazyswarm2 #244: Issue running crazyflie_examples scripts in simulation backend

Shyamyar, 2023-04-11:
I am just beginning to learn to work with crazyflies and ROS2. I am working in WSL right now and not an exclusive Linux system. I was going through the steps of "[Usage](https://imrclab.github.io/crazyswarm2/usage.html)" in the documentation for "[Crazyswarm2](https://imrclab.github.io/crazyswarm2/index.html)" and tried to run the hello_world script with the simulation backend using the line of codes mentioned, but it issued the following error. It shows the same error for all the scripts in crazyflie_examples when run with the sim backend.

```
username@laptop_name:~$ ros2 run crazyflie_examples hello_world --ros-args -p use_sim_time:=True
Traceback (most recent call last):
  File "/home/username/ros2_ws/install/crazyflie_examples/lib/crazyflie_examples/hello_world", line 33, in <module>
    sys.exit(load_entry_point('crazyflie-examples==2.0.0', 'console_scripts', 'hello_world')())
  File "/home/username/ros2_ws/install/crazyflie_examples/local/lib/python3.10/dist-packages/crazyflie_examples/hello_world.py", line 12, in main
    swarm = Crazyswarm()
  File "/home/username/ros2_ws/install/crazyflie_py/lib/python3.10/site-packages/crazyflie_py/crazyswarm_py.py", line 11, in __init__
    self.allcfs = CrazyflieServer()
  File "/home/username/ros2_ws/install/crazyflie_py/lib/python3.10/site-packages/crazyflie_py/crazyflie.py", line 722, in __init__
    self.paramTypeDict = allParamTypeDicts["all"]
KeyError: 'all'
[ros2run]: Process exited with failure 1
```

I am not sure why so because I have run a couple of physical experiments with services like takeoff, landing, goto, and using keyboard inputs for simple motion. I would be glad if anyone can assist me here. I am learning to write scripts for controlling crazyflies such that they can avoid collisions. Before that, I would like to test the scripts in the simulation environment.

whoenig, 2023-04-11:
Did you pull recently? In particular, please make sure you have c1e5b5c232cef3f8da8e5efb400651c6a8560041.

Shyamyar, 2023-04-11:
Thanks so much. That worked. I also needed to rebuild the ros_ws package with `colcon build` though.

closed 2023-04-12 by knmcguire