IMRCLab/crazyswarm2 #245: Prepare for Crazyradio 2

knmcguire, 2023-04-19:
[Crazyradio 2](https://www.bitcraze.io/products/crazyradio-2-0/) is now out! However, there might be some things we need to be aware of in regard of it's version. I had one time that I had an error from crazyflie-tools that it said that the version wasn't right but I was not able to replicate this anymore. 

The version that should be given by Crazyradio 2 is '3.2', so maybe just to be sure we can place an exception?

williamleong, 2023-07-07:
The error message shown is:

````
[crazyflie_server-3] terminate called after throwing an instance of 'std::runtime_error'
[crazyflie_server-3]   what():  Issue with connection radiobroadcast://*/80/2M. Your radio with firmware 3.2 does not support broadcast communication. Please upgrade your Crazyradio firmware!
````

knmcguire, 2023-07-26:
It's a very dirty fix but if it doesn't work yet, one can comment out this part of one of the submodules, until a more proper fix has been made

https://github.com/bitcraze/crazyflie-link-cpp/blob/fcde6a0850f2f3127fb5f65b76aaabda530d7cd9/src/CrazyradioThread.cpp#L198-L203

whoenig, 2023-07-27:
Duplicate of https://github.com/IMRCLab/crazyswarm2/issues/264. Let's continue the discussion there.

closed 2023-07-27 by whoenig