IMRCLab/crazyswarm2 #246: Why is cmdVel commented out?

DS3a, 2023-04-20:
In the file [crazyflie.py](https://github.com/IMRCLab/crazyswarm2/blob/main/crazyflie_py/crazyflie_py/crazyflie.py), I noticed that `cmdVel` is commented out, which is a feature which I was looking for, are there any issues in its integration with ros2? The function is available on crazyswarm, and is widely used for low level control

whoenig, 2023-04-22:
I don't think there is any issue, we just haven't gotten around in testing it.

DS3a, 2023-04-30:
Oh, right, I will try to test it and create a PR

matthew-william-lock, 2023-10-05:
@DS3a Have you submitted a PR?

matthew-william-lock, 2023-10-10:
@whoenig @knmcguire 

Regarding the implementation of the cmdVel functionality, would something like this work?

In ```crazyflie_sim/crazyflie_sim/crazyflie_server.py```, creating the following topic callback
```python
_cmd_vel_legacy_changed
def _cmd_vel_legacy_changed(self, msg : Twist, name=""):
    """
    Topic update callback to control the attitude and thrust
        of the crazyflie with teleop
    """
    roll = msg.linear.y
    pitch = -msg.linear.x
    yawrate = msg.angular.z
    thrust = int(min(max(msg.linear.z, 0, 0), 60000))
    self.cfs[name].cmdVel(roll, pitch, yawrate, thrust)
```

and then adding ```cmdVel``` to ```crazyflie_sim/crazyflie_sim/crazyflie_sil.py```
```python
 def cmdVel(self, roll, pitch, yawrate, thrust):
    self.mode = CrazyflieSIL.MODE_LOW_VELOCITY
    self.setpoint.attitudeRate.roll = np.degrees(roll)
    self.setpoint.attitudeRate.pitch = np.degrees(pitch)
    self.setpoint.attitudeRate.yaw = np.degrees(yawrate)
    self.setpoint.thrust = thrust
    self.setpoint.mode.x = firm.modeDisable
    self.setpoint.mode.y = firm.modeDisable
    self.setpoint.mode.z = firm.modeDisable
    self.setpoint.mode.roll = firm.modeAbs
    self.setpoint.mode.pitch = firm.modeAbs
    self.setpoint.mode.yaw = firm.modeAbs
    self.setpoint.mode.quat = firm.modeDisable
```

   
Similar to https://github.com/IMRCLab/crazyswarm2/blob/379601ea4c5f98f82c96749bfbe8e82a458ea9f9/crazyflie/scripts/crazyflie_server.py#L825

My main concern is the setpoint mode as I am not exactly sure which options to select to mimic the on-board controllers.

Cheers!

DS3a, 2023-10-11:
I'm sorry, I wasn't able to find the time to work on this

matthew-william-lock, 2023-10-11:
I have updated ```cmdVel``` to 
```python
def cmdVel(self, roll, pitch, yawrate, thrust):
    self.mode = CrazyflieSIL.MODE_LOW_VELOCITY
    self.setpoint.attitude.roll     = np.degrees(roll)
    self.setpoint.attitude.pitch    = np.degrees(pitch)
    self.setpoint.attitudeRate.yaw  = np.degrees(yawrate)
    self.setpoint.thrust            = thrust
    
    self.setpoint.mode.x        = firm.modeDisable
    self.setpoint.mode.y        = firm.modeDisable
    self.setpoint.mode.z        = firm.modeDisable
    self.setpoint.mode.roll     = firm.modeAbs
    self.setpoint.mode.pitch    = firm.modeAbs
    self.setpoint.mode.yaw      = firm.modeVelocity
    self.setpoint.mode.quat     = firm.modeDisable
```

but still see no movement in the sim backend (testing by publishing message with maximum thrust and zero pitch, roll and yawrate).

matthew-william-lock, 2023-10-11:
> I have updated `cmdVel` to
>
> ```python
> def cmdVel(self, roll, pitch, yawrate, thrust):
>     self.mode = CrazyflieSIL.MODE_LOW_VELOCITY
>     self.setpoint.attitude.roll     = np.degrees(roll) [...]

Nevermind, there was a mistake in my sim, this seems to work 😄

whoenig, 2023-10-12:
Great - PRs are welcome:-)

niruem, 2026-01-13:
@mwlock What mistake did you have in your sim? Because I have a similar issue as you. My RPY commands don't go through even if I hard-code the values in the CmdVel() function within the crazyflie_sil.py file. However, my thrust does seem to work.