IMRCLab/crazyswarm2 #254: Fatal error no such file or directory, (while trying to make python-bindings)

TomoCoder123, 2023-06-01:
While trying to make the python bindings, I come across this error. Is there a dependency that I do not have installed? I have tried the solutions offered in #140. Using ROS2 Galactic and Ubuntu 20.04.

[image]

whoenig, 2023-06-01:
You probably didn't clone recursively, i.e., you need to init and update your submodules: `git submodules update --init`.

TomoCoder123, 2023-06-01:
`Hm, for some reason when I ran that command, it still didn't add the DSP/source file path. I manually placed the DSP/source folder and that seemed to fix it.

closed 2023-06-01 by TomoCoder123