IMRCLab/crazyswarm2 #260: Crazyswarm2 fails to connect to crazyflie when launching launch.py

AnassQach, 2023-07-07:
I'm using ROS2-Galactic, Crazyswarm2 on Ubuntu 20.04.
The Cfclient has no problem connecting to the crazyflie. See Screenshot:
[image]

But when i launch the Crazyswarm sim, the cf doesn't seem to appear on rviz.

[image]

[image]

Any help or tipp will be much appreciated. I'm kinda new to this.

my **crazyflies.yaml**:

++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
# named list of all robots
robots:
  cf231:
    enabled: false
    uri: radio://0/80/2M/E7E7E7E7E7
    initial_position: [0, 0, 0]
    type: cf21  # see robot_types
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    # firmware_logging:
    #   enabled: true
    #   custom_topics:
    #     topic_name3: 
    #       frequency: 1
    #       vars: ["acc.x", "acc.y"]
  cf5:
    enabled: true
    uri: radio://0/80/2M/E7E7E7E705
    initial_position: [0, 0, 0]
    type: cf21  # see robot_types
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    #firmware_logging:
    #  custom_topics:
    #    topic_name3: 
    #      frequency: 1
    #      vars: ["acc.x", "acc.y", "acc.z"]
# Definition of the various robot types
robot_types:
  cf21:
    motion_capture:
      enabled: true
      # only if enabled; see motion_capture.yaml
      marker: default_single_marker
      dynamics: default
    big_quad: false
    battery:
      voltage_warning: 3.8  # V
      voltage_critical: 3.7 # V
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    #firmware_logging:
    #   enabled: true
    #   default_topics:
    #   pose:
    #     frequency: 1 # Hz
    #   custom_topics:
    #    topic_name3: 
    #      frequency: 1
    #      vars: ["acc.x", "acc.y", "acc.z"]

  cf21_mocap_deck:
    motion_capture:
      enabled: true
      # only if enabled; see motion_capture.yaml
      marker: mocap_deck
      dynamics: default
    big_quad: false
    battery:
      voltage_warning: 3.8  # V
      voltage_critical: 3.7 # V

    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5

# global settings for all robots
all:
  # firmware logging for all drones (use robot_types/type_name to set per type, or
  # robots/drone_name to set per drone)
  firmware_logging:
    enabled: false
    default_topics:
     # remove to disable default topic
      pose:
        frequency: 10 # Hz
    #custom_topics:
    #  topic_name1:
    #    frequency: 10 # Hz
    #    vars: ["stateEstimateZ.x", "stateEstimateZ.y", "stateEstimateZ.z", "pm.vbat"]
    #  topic_name2:
    #    frequency: 1 # Hz
    #    vars: ["stabilizer.roll", "stabilizer.pitch", "stabilizer.yaw"]
  # firmware parameters for all drones (use robot_types/type_name to set per type, or
  # robots/drone_name to set per drone)
  firmware_params:
    commander:
      enHighLevel: 1
    stabilizer:
      estimator: 2 # 1: complementary, 2: kalman
      controller: 2 # 1: PID, 2: mellinger
    # ring:
    #   effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
    #   solidBlue: 255 # if set to solid color
    #   solidGreen: 0 # if set to solid color
    #   solidRed: 0 # if set to solid color
    #   headlightEnable: 0
    locSrv:
      extPosStdDev: 1e-3
      extQuatStdDev: 0.5e-1
    # kalman:
    #   resetEstimation: 1
  broadcasts:
    num_repeats: 15 # number of times broadcast commands are repeated
    delay_between_repeats_ms: 1 # delay in milliseconds between individual repeats

++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

whoenig, 2023-07-09:
You can connect to your drone, but your setting is to track it using a motion capture system. However, you do not receive data from your motion capture. Perhaps check your setting in motion_capture.yaml, or provide some background on how you want to track your drone so we can give some more guidance on what parameters to adjust.

AnassQach, 2023-07-10:
Thank you for the response, i'm not using a motion capture system. Only Drone, Flowdeck and Radio dongle. 
But i fixed the problem. my firmware logging was disabled. Now it works,  i can see the cf in rviz.

whoenig, 2023-07-10:
Great - thanks for your feedback.

closed 2023-07-10 by whoenig

HWMathSEU, 2024-05-06:
> Thank you for the response, i'm not using a motion capture system. Only Drone, Flowdeck and Radio dongle. But i fixed the problem. my firmware logging was disabled. Now it works, i can see the cf in rviz.

Hi Whoening, I currently meet the same problem. Can you explain more details? Thank you very much