IMRCLab/crazyswarm2 #263: Does not work with 4 crazyradios

Kaijun101, 2023-07-17:
Hi, 

I have used 4 crazyradios to connect to 20 drones, but noticed that there is always this error that say no crazyradio with ID3 found. This error appears when the crazyradio ID is more than 2. Can you kindly advise me on this error?

whoenig, 2023-07-17:
Can you post the output of `lsusb`?

Kaijun101, 2023-07-17:
lsusb:

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 06cb:00f9 Synaptics, Inc. 
Bus 003 Device 003: ID 13d3:5271 IMC Networks Integrated RGB Camera
Bus 003 Device 060: ID 1915:7777 Nordic Semiconductor ASA Crazyradio PA USB Dongle
Bus 003 Device 058: ID 1915:7777 Nordic Semiconductor ASA 
Bus 003 Device 057: ID 1915:7777 Nordic Semiconductor ASA Crazyradio PA USB Dongle
Bus 003 Device 056: ID 1a40:0801 Terminus Technology Inc. USB 2.0 Hub
Bus 003 Device 007: ID 05e3:0751 Genesys Logic, Inc. microSD Card Reader
Bus 003 Device 005: ID 1a40:0801 Terminus Technology Inc. USB2.0 Hub             
Bus 003 Device 002: ID 2109:2817 VIA Labs, Inc. USB2.0 Hub             
Bus 003 Device 008: ID 8087:0032 Intel Corp. 
Bus 003 Device 006: ID 056a:52a3 Wacom Co., Ltd Pen and multitouch sensor
Bus 003 Device 059: ID 1915:7777 Nordic Semiconductor ASA Crazyradio PA USB Dongle
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 002 Device 002: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

dmesg:
[ 2806.928547] usb 3-5: new high-speed USB device number 56 using xhci_hcd
[ 2807.076956] usb 3-5: New USB device found, idVendor=1a40, idProduct=0801, bcdDevice= 1.00
[ 2807.076966] usb 3-5: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 2807.076970] usb 3-5: Product: USB 2.0 Hub
[ 2807.078448] hub 3-5:1.0: USB hub found
[ 2807.078507] hub 3-5:1.0: 4 ports detected
[ 2807.388590] usb 3-5.1: new full-speed USB device number 57 using xhci_hcd
[ 2807.513222] usb 3-5.1: New USB device found, idVendor=1915, idProduct=7777, bcdDevice=99.55
[ 2807.513226] usb 3-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=29
[ 2807.513228] usb 3-5.1: Product: Crazyradio PA USB Dongle
[ 2807.513228] usb 3-5.1: Manufacturer: Bitcraze
[ 2807.513229] usb 3-5.1: SerialNumber: 3AD6093972
[ 2807.600562] usb 3-5.2: new full-speed USB device number 58 using xhci_hcd
[ 2807.728054] usb 3-5.2: New USB device found, idVendor=1915, idProduct=7777, bcdDevice=99.55
[ 2807.728062] usb 3-5.2: New USB device strings: Mfr=1, Product=2, SerialNumber=29
[ 2807.728064] usb 3-5.2: Product: Crazyradio PA USB Dongle
[ 2807.728066] usb 3-5.2: Manufacturer: Bitcraze
[ 2807.728067] usb 3-5.2: SerialNumber: EA181A292F
[ 2809.608536] usb 3-1: new full-speed USB device number 59 using xhci_hcd
[ 2809.759849] usb 3-1: New USB device found, idVendor=1915, idProduct=7777, bcdDevice=99.55
[ 2809.759852] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=29
[ 2809.759853] usb 3-1: Product: Crazyradio PA USB Dongle
[ 2809.759854] usb 3-1: Manufacturer: Bitcraze
[ 2809.759855] usb 3-1: SerialNumber: 21E7A98056
[ 2812.288507] usb 3-6: new full-speed USB device number 60 using xhci_hcd
[ 2812.507810] usb 3-6: New USB device found, idVendor=1915, idProduct=7777, bcdDevice=99.55
[ 2812.507813] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=29
[ 2812.507814] usb 3-6: Product: Crazyradio PA USB Dongle
[ 2812.507815] usb 3-6: Manufacturer: Bitcraze
[ 2812.507815] usb 3-6: SerialNumber: 6CEF0A76B2

Kaijun101, 2023-07-17:
The system still only registered 3 crazyradio only. Below is the error message:

crazyflie_server-3] added new 21E7A98056 0                                                           │
[crazyflie_server-3] added new 6CEF0A76B2 1                                                           │
[crazyflie_server-3] added new 3AD6093972 2                                                           │
[crazyflie_server-3] terminate called after throwing an instance of 'std::runtime_error'              │
[crazyflie_server-3]   what():  No Crazyradio with id=3 found.There are 3 Crazyradios connected.      │

whoenig, 2023-07-17:
Strange! Does the radio `EA181A292F` work by itself?

Kaijun101, 2023-07-17:
Yes, the dongle works by itself. Seems like the fourth crazyradio is unable to detect.

williamleong, 2023-07-17:
From the above:
```
Bus 003 Device 060: ID 1915:7777 Nordic Semiconductor ASA Crazyradio PA USB Dongle
Bus 003 Device 058: ID 1915:7777 Nordic Semiconductor ASA
Bus 003 Device 057: ID 1915:7777 Nordic Semiconductor ASA Crazyradio PA USB Dongle
Bus 003 Device 059: ID 1915:7777 Nordic Semiconductor ASA Crazyradio PA USB Dongle
```

Device 058 doesn't seem to be detected correctly as the name is "Nordic Semiconductor ASA" instead of "Nordic Semiconductor ASA Crazyradio PA USB Dongle"

```
[ 2807.600562] usb 3-5.2: new full-speed USB device number 58 using xhci_hcd
[ 2807.728054] usb 3-5.2: New USB device found, idVendor=1915, idProduct=7777, bcdDevice=99.55
[ 2807.728062] usb 3-5.2: New USB device strings: Mfr=1, Product=2, SerialNumber=29
[ 2807.728064] usb 3-5.2: Product: Crazyradio PA USB Dongle
[ 2807.728066] usb 3-5.2: Manufacturer: Bitcraze
[ 2807.728067] usb 3-5.2: SerialNumber: EA181A292F
```

It is also the one that is not detected.

williamleong, 2023-07-17:
Fixed the above with https://askubuntu.com/questions/1397139/usb-devices-often-named-incorrectly so all are showing "Nordic Semiconductor ASA Crazyradio PA USB Dongle" now.

But still not always able to detect 4 radios. The radio that is not detected will have its red LED constantly on.

whoenig, 2023-07-19:
Sounds like the naming issue is a bug of lsusb, not of libusb (which we directly rely on). If they are all on the same USB hub, does it help to perhaps put 2 on one hub and 2 on another hub (connected to a different USB port on the computer)?

williamleong, 2023-07-19:
We have tried a few combinations, such as 3 radios on a USB hub and 1 radio directly connected, or all 4 directly connected, but it can still happen.

whoenig, 2023-07-19:
This almost sounds more like an issue of libusb, rather than crazyflie-link-cpp. Perhaps upgrading your OS (and thus libusb) will improve the situation.

williamleong, 2023-07-20:
Thanks, will give it a try.

williamleong, 2023-07-20:
Isn't libusb linked to crazyflie-link-cpp as a submodule?

whoenig, 2023-07-20:
Yes, but only for the Windows build, on Linux it uses the system version, see https://github.com/bitcraze/crazyflie-link-cpp/blob/master/CMakeLists.txt#L31-L53. Might be easy to switch to a newer version by removing the if on Linux, but I have not tried that.

williamleong, 2023-07-20:
I see, thanks. I'll give it a try.

whoenig, 2023-07-28:
Very strange. I had another look at the code and detecting radios is done at the very beginning, even before any threads should have been started. As you noted in the other discussion, there might be a race condition in the code.

williamleong, 2023-07-29:
Yes this is strange to me as well. The race condition should only stop data from being sent or received. This is something different. It could be the safelink firmware bug you are thinking of.

What I suspect is that repeat calls to the radio's send and receive API without any delay might trigger some problem that causes the firmware to be stuck. Once it is stuck, the red LED remains lit. Even if we terminate Crazyswarm2, the red LED remains lit and subsequently, Crazyswarm2 or any other application (reboot/battery) cannot detect the radio anymore.

This does not happen with the cflib python library, so maybe it is slower and does not cause the problem as you suggested, or it is because broadcast is not used with cflib.