IMRCLab/crazyswarm2 #264: Error with crazyradio2.0

Kaijun101, 2023-07-17:
Hi,

When I used crazyradio2, I received the error: LIBUSB_ERROR_TIMEOUT.
Can you advise on this error?

whoenig, 2023-07-17:
FYI: We don't have a Crazyradio2 in the lab yet, so this hasn't been tested on our side.

Kaijun101, 2023-07-17:
Oh thanks a lot for the info! :)

Kaijun101, 2023-07-20:
Hi @whoenig, since now crazyradio2.0 is the de facto radio for the crazyflie system, will there be plans to support this as well?

whoenig, 2023-07-20:
Yes of course. I think Kimberly did some brief tests and it should already work with the cflib backend. We'll order some this week and test it with the (default) cpp backend.

Kaijun101, 2023-07-21:
Cool, do let me know of the test. I'm curious to see if there is the error mentioned above.

knmcguire, 2023-07-26:
Hi! I've kinda started an issue about this already a while ago: https://github.com/IMRCLab/crazyswarm2/issues/245. It contains a quick and dirty fix if you are in a hurry, or else we can wait for wolfgang

whoenig, 2023-07-27:
I have the same issue as @Kaijun101. Broadcast itself seems to work (verified by running the broadcast_example in crazyflie_link_cpp) standalone. The error comes from crazyflie_link_cpp and it looks like it might be caused by having a connection without broadcast and one with broadcasts open at the same time (which was no issue with the Crazyradio PA).

e0538382, 2023-09-28:
Currently, I have also been trying to get CrazySwarm2 to work with CrazyRadio2 and face problems running the provided examples hello_world  where the version of the crazyradio2 firmware is rejected. I tried to fix it by commenting out the version control block: #245. However, the same error message still pops up as seen in the log below. 

May I ask if there is any alternative fix for this or did we miss any step for building the file (colcon build --symlink-install)?

[crazyflie_server-4] terminate called after throwing an instance of 'std::runtime_error'
[crazyflie_server-4] what(): Issue with connection radiobroadcast://*/80/2M. Your radio with firmware 3.2 does not support broadcast communication. Please upgrade your Crazyradio firmware!
[ERROR] [crazyflie_server-4]: process has died [pid 15020, exit code -6, cmd '/home/ryan/ros2_ws/install/crazyflie/lib/crazyflie/crazyflie_server --ros-args -r __node:=crazyflie_server --params-file /tmp/launch_params_x0cf7z3c --params-file /tmp/launch_params_jtlmd2_1'].

One more thing is that I also tried the teleop example and although no error message pops up, despite us sending in the input, the crazyflie does not seem to respond to it.

whoenig, 2023-09-28:
Some of the submodules weren't properly updated - I just pushed a fix. Perhaps you can update (see https://imrclab.github.io/crazyswarm2/installation.html#updating) and try again? Note that you need the latest crazyradio2 firmware from https://github.com/bitcraze/crazyflie-link-cpp/files/12429056/crazyradio2-CRPA-emulation.zip (see https://github.com/bitcraze/crazyflie-link-cpp/issues/37).

e0538382, 2023-10-03:
Thank you so much! I have managed to run the hello_world and figure8 examples after updating, will let you know if I face any other problems

whoenig, 2023-10-03:
Great - thanks for the confirmation. I think with that we can close this issue.

closed 2023-10-03 by whoenig