IMRCLab/crazyswarm2 #270: Error in building ROS2 workspace

gaurav713914, 2023-08-05:
While running the `colcon build --symlink-install` command we are getting the following error:

```
Starting >>> crazyflie_interfaces
Starting >>> motion_capture_tracking_interfaces
--- stderr: crazyflie_interfaces                                         
CMake Error at /opt/ros/galactic/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:60 (message):
  execute_process(/home/mocaplab/anaconda3/bin/python3 -m rosidl_adapter
  --package-name crazyflie_interfaces --arguments-file
  /home/mocaplab/ros2_ws/build/crazyflie_interfaces/rosidl_adapter__arguments__crazyflie_interfaces.json
  --output-dir
  /home/mocaplab/ros2_ws/build/crazyflie_interfaces/rosidl_adapter/crazyflie_interfaces
  --output-file
  /home/mocaplab/ros2_ws/build/crazyflie_interfaces/rosidl_adapter/crazyflie_interfaces.idls)
  returned error code 1:

  Traceback (most recent call last):

    File "/home/mocaplab/anaconda3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
      "__main__", mod_spec)
    File "/home/mocaplab/anaconda3/lib/python3.7/runpy.py", line 85, in _run_code
      exec(code, run_globals)
    File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_adapter/main.py", line 55, in main
      pathlib.Path(relative_path), output_dir)
    File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_adapter/__init__.py", line 18, in convert_to_idl
      from rosidl_adapter.msg import convert_msg_to_idl
    File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_adapter/msg/__init__.py", line 16, in <module>
      from rosidl_adapter.resource import expand_template
    File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_adapter/resource/__init__.py", line 19, in <module>
      import em

  ModuleNotFoundError: No module named 'em'

Call Stack (most recent call first):
  /opt/ros/galactic/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:20 (rosidl_generate_interfaces)

---
Failed   <<< crazyflie_interfaces [0.27s, exited with code 1]
Aborted  <<< motion_capture_tracking_interfaces [0.26s]

Summary: 0 packages finished [0.42s]
  1 package failed: crazyflie_interfaces
  1 package aborted: motion_capture_tracking_interfaces
  2 packages had stderr output: crazyflie_interfaces motion_capture_tracking_interfaces
  5 packages not processed
```

knmcguire, 2023-08-07:
Hi! This seems to be a ROS2-specific issue... have you tried searching on their support platform for this? https://robotics.stackexchange.com/. Maybe updating ROS2 packages will resolve a lot a s well.

gaurav713914, 2023-08-07:
Thank you for your suggestion, it is building now but its still showing error with 3 packages and the error is given below:

```
`Starting >>> crazyflie_interfaces
Starting >>> motion_capture_tracking_interfaces
Finished <<< motion_capture_tracking_interfaces [0.40s]                       
Starting >>> motion_capture_tracking
Finished <<< motion_capture_tracking [0.16s]                               
Finished <<< crazyflie_interfaces [0.65s]                  
Starting >>> crazyflie_py
Starting >>> crazyflie
Starting >>> crazyflie_sim
Finished <<< crazyflie [0.25s]                                                
--- stderr: crazyflie_sim                                  
/home/mocaplab/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< crazyflie_sim [0.53s]
--- stderr: crazyflie_py                   
/home/mocaplab/.local/lib/python3.10/site-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
/home/mocaplab/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< crazyflie_py [0.90s]
Starting >>> crazyflie_examples
--- stderr: crazyflie_examples                           
/home/mocaplab/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< crazyflie_examples [0.39s]

Summary: 7 packages finished [2.08s]
  3 packages had stderr output: crazyflie_examples crazyflie_py crazyflie_sim
`
```

Please guide regarding this.

gaurav713914, 2023-08-08:
Thank you for your response, I am closing this issue as completed. However we are facing other problems for which I am opening other issue. Thanks again!

closed 2023-08-08 by gaurav713914