IMRCLab/crazyswarm2 #271: Unable to run hello_world.py

gaurav713914, 2023-08-08:
We are unable to run hello_world.py or any other example code in crazyflie_examples package. However the launch.py is running perfectly with both single marker as well as an active deck mounted on top of crazyflie and we are able to see the crazyflie in Rviz. Please help regarding the same, we are using Qualisys motion capture system. We have Ubuntu 22.04 and ROS humble. When we run hello_world.py we don't see any change in the position of crazyflie, we also don't see it connecting with the system because it's yellow LED does not blink. The output of the terminal is also blank after running it.

whoenig, 2023-08-08:
We would need to see the console output of the launch file, in particular it should print something if it tries to send the takeoff command.

gaurav713914, 2023-08-08:
We are getting the following output in the terminal in which we ran the launch.py 

```
mocaplab@mocaplab-HP-Z1-G8-Tower-Desktop-PC:~/ros2_ws$ ros2 launch crazyflie launch.py
[INFO] [launch]: All log files can be found below /home/mocaplab/.ros/log/2023-08-09-00-16-36-420291-mocaplab-HP-Z1-G8-Tower-Desktop-PC-6244
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [motion_capture_tracking_node-1]: process started with pid [6246]
[INFO] [teleop-2]: process started with pid [6248]
[INFO] [joy_node-3]: process started with 
![Screenshot from 2023-08-09 00-20-23](https://github.com/IMRCLab/crazyswarm2/assets/131975334/681530fe-13e3-4abf-8140-754bfecbff60)
pid [6250]
[INFO] [crazyflie_server-4]: process started with pid [6252]
[INFO] [rviz2-5]: process started with pid [6254]
[rviz2-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[teleop-2] [INFO] [1691520396.567468463] [teleop]: Mode changed to cmd_vel_world
[motion_capture_tracking_node-1] [WARN] [1691520396.600939356] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.600860 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.610533365] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.610505 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.619653444] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.619615 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.630375727] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.630321 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.639726283] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.639678 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.649643143] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.649572 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.660566357] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.660520 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.675583372] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.675508 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.680616401] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.680561 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.689457680] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.689419 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.699457461] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.699432 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.709567586] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.709535 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.719534337] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.719499 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.729433695] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.729408 s.
[rviz2-5] [INFO] [1691520396.737128040] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1691520396.737251988] [rviz2]: OpenGl version: 4.3 (GLSL 4.3)
[motion_capture_tracking_node-1] [WARN] [1691520396.739803420] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.739754 s.
[rviz2-5] [INFO] [1691520396.745537308] [rviz2]: Stereo is NOT SUPPORTED
[motion_capture_tracking_node-1] [WARN] [1691520396.749620621] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.749547 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.759363407] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.759332 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.769729201] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.769665 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.779753442] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.779716 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.791306979] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.791256 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.799594953] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.799521 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.809579575] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.809496 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.819958794] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.819890 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.829838601] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.829772 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.839673145] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.839624 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.849721449] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.849560 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.859965466] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.859883 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.872426810] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.872356 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.880644586] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.880574 s.
[motion_capture_tracking_node-1] [WARN] [1691520396.914988311] [motion_capture_tracking]: No updated pose for cf231 for 1691520396.914894 s.
[... cut ...]
[ERROR] [crazyflie_server-4]: process[crazyflie_server-4] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
[INFO] [crazyflie_server-4]: sending signal 'SIGKILL' to process[crazyflie_server-4]
[ERROR] [crazyflie_server-4]: process has died [pid 6252, exit code -9, cmd '/home/mocaplab/ros2_ws/install/crazyflie/lib/crazyflie/crazyflie_server --ros-args -r __node:=crazyflie_server --params-file /tmp/launch_params_08_d2iwz --params-file /tmp/launch_params_d2dl9sgy'].
mocaplab@mocaplab-HP-Z1-G8-Tower-Desktop-PC:~/ros2_ws$ 

```

whoenig, 2023-08-08:
- The no updated pose messages are not normal.
- It doesn't look like your hello_world script tried to initiate takeoff. Do you have a joystick and does that work (start button on an xbox360 compatible joystick)? If not, there is also a commandline to initiate takeoff (see documentation).

gaurav713914, 2023-08-09:
We tried with the joystick of Xbox compatible controller and the following is the part of the output (for launch.py) in terminal after pressing start button on controller.

`[teleop-2] [ERROR] [1691588687.740166540] [teleop]: Takeoff service not ready!
[motion_capture_tracking_node-1] [WARN] [1691588687.740847480] [motion_capture_tracking]: No updated pose for cf231 for 38.115540 s.`

When we are trying with the take off commands we are getting the following:

`ocaplab@mocaplab-HP-Z1-G8-Tower-Desktop-PC:~/ros2_ws$ ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E701 --mode sysoff
mocaplab@mocaplab-HP-Z1-G8-Tower-Desktop-PC:~/ros2_ws$ ros2 param set crazyflie_server cf231.params.commander.enHighLevel 1
Node not found
mocaplab@mocaplab-HP-Z1-G8-Tower-Desktop-PC:~/ros2_ws$ ros2 param set crazyflie_server cf1.params.stabilizer.estimator 2
Node not found
mocaplab@mocaplab-HP-Z1-G8-Tower-Desktop-PC:~/ros2_ws$ ros2 service call cf1/takeoff crazyflie_interfaces/srv/Takeoff "{height: 0.5, duration: {sec: 2}}"
waiting for service to become available...`

We waited for about 5 minutes. The motion capture system is running.

We also tried with the `ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E701 --mode syson` and still getting the same.

whoenig, 2023-08-09:
- ros2 run crazyflie reboot only works if the server is not running
- All other commands only work if the server is running
- The name of your cf is cf231 according to your yaml file, so you need to adjust your commands accordingly. Please also double check the remapping in your launch file, since you might be waiting for a Crazyflie that doesn't even exist in your set up.

gaurav713914, 2023-08-10:
We tried to move the cf manually and got the following output on the terminal.

```
[motion_capture_tracking_node-1] [WARN] [1691674218.762064921] [motion_capture_tracking]: Dynamic check failed for rigidBody cf231
[motion_capture_tracking_node-1] wroll: -45.0463 >= 20
[motion_capture_tracking_node-1] wyaw: -36.1802 >= 10
[motion_capture_tracking_node-1] roll: 1.81434 >= 1.4
[motion_capture_tracking_node-1] 

```

```
[motion_capture_tracking_node-1] [WARN] [1691674219.032144107] [motion_capture_tracking]: No updated pose for cf231 for 0.140112 s.
[motion_capture_tracking_node-1] [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
[motion_capture_tracking_node-1] [WARN] [1691674219.053218331] [motion_capture_tracking]: ICP did not converge! for rigidBody cf231
[motion_capture_tracking_node-1] [WARN] [1691674219.053274645] [motion_capture_tracking]: No updated pose for cf231 for 0.010917 s.
[motion_capture_tracking_node-1] [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
[motion_capture_tracking_node-1] [WARN] [1691674219.072651359] [motion_capture_tracking]: ICP did not converge! for rigidBody cf231
[motion_capture_tracking_node-1] [WARN] [1691674219.072795640] [motion_capture_tracking]: No updated pose for cf231 for 0.010135 s.
[motion_capture_tracking_node-1] [WARN] [1691674219.082169053] [motion_capture_tracking]: No updated pose for cf231 for 0.019872 s.

```

However, after some time the output on the terminal again changed to `No updated Pose` even when the position of cf was changed manually (could be seen in RVIZ)

The name of the crazyflie is same in the launch file (crayflies.yaml) and QTM that is cf231 and we also tried changing it to cf1 in both qtm as well as crazyflies.yaml but still same issue.

Do we have to configure any other files?
We haven't changes any files other than crayflies.yaml and motion_capture.yaml

We have cross checked every name and address of URI at least two times, please help.

Also, how to check if the server is running for crazyswarm2 and how to run it if its not running?

whoenig, 2023-08-10:
How is the qualisys computer with QTM connected to the Ubuntu machine? Perhaps try running the example from https://github.com/IMRCLab/libmotioncapture to see if you actually do get a reliable connection. Generally, you want the two computers to be on the same ethernet (no wifi!).

For the joystick, you have to update the remapping in https://github.com/IMRCLab/crazyswarm2/blob/main/crazyflie/launch/launch.py#L78-L83.

The server is running whenever you use ros2 launch (see https://github.com/IMRCLab/crazyswarm2/blob/main/crazyflie/launch/launch.py#L100-L107). You can double check with rqt_graph to see which nodes are active.

push2312, 2023-08-10:
The computer running QTM is on Wifi (static IP address) and the Ubuntu is connected to ethernet.

We are able to see the pose of the crazyflie in RVIZ. Is it possible that the RVIZ shows the crazyflie pose in unstable connection?

We will try the mentioned solutions and let you know.

whoenig, 2023-08-11:
Rviz is not a great indicator because it caches old data without indication. So it's possible everything looks fine in rviz, but you actually have a very high latency.

gaurav713914, 2023-08-11:
Oh, thanks for clarification about Rviz. The problem was in the firmware of crazyflie, when we were using an older firmware then the hello_world was not working. We tried different versions of firmware and found 2/2023 version to be most suitable as both hello_world and joystick control are working otherwise only joystick control was working in 12/2022 version and hello_world was not working. We also changed our network configuration to bring both the PCs on one router using ethernet cables so it was the bigger issue that we did not know because it was mentioned nowhere. Thanks for all the support on the same.

However, we are still not able to run multiple crazyflies using the active marker deck because the QTM does not allow two or more crazyflies to be made rigid bodies at once when we use active marker deck. That I think is the issue from QTM side so I am closing this issue. Thanks again!

closed 2023-08-11 by gaurav713914