IMRCLab/crazyswarm2 #274: Issue with Instability in rivz_crazyflie (backend:=cflib)

eunseo01, 2023-08-09:
We are currently using the cflib backend and 'optitrack' motion capture system for our setup. Although we can visualize the Crazyflie drone in RViz, we are encountering repetitive errors similar to what is captured in the screenshots. We have already completed modifications to the YAML files and successfully installed the crazyflie-lib-python library. As a precaution, we have also reconfigured the origin and orientation of the motion capture system to ensure accuracy.

We are seeking advice regarding any potential misconfigurations or issues that might arise when executing the command ros2 launch crazyflie launch.py backend:=cflib.

Additionally, we have another inquiry. During our physical experiments, we came across the command ros2 param set crazyflie_server cf2.params.commander.enHighLevel 1. However, we noticed that the parameter 'cf2.params.commander.enHighLevel' does not seem to be present within the crazyflie_server package. We are wondering if we might have missed a step or if there is additional information required to properly address this parameter.

For your convenience, we have attached images depicting our current setup and the encountered errors. We appreciate any guidance or assistance you can provide."

[image]

1.crazyflies.yaml
`# named list of all robots
robots:
  cf2:
    enabled: true
    uri: radio://0/80/2M/E7E7E7E702
    initial_position: [0, 0, 0]
    type: cf21  # see robot_types
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    # firmware_logging:
    #   enabled: true
    #   custom_topics:
    #     topic_name3: 
    #       frequency: 1
    #       vars: ["acc.x", "acc.y"]
  cf5:
    enabled: false
    uri: radio://0/80/2M/E7E7E7E705
    initial_position: [0, -0.5, 0]
    type: cf21  # see robot_types
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    #firmware_logging:
    #  custom_topics:
    #    topic_name3: 
    #      frequency: 1
    #      vars: ["acc.x", "acc.y", "acc.z"]
# Definition of the various robot types
robot_types:
  cf21:
    motion_capture:
      enabled: true
      # only if enabled; see motion_capture.yaml
      marker: default
      dynamics: default
    big_quad: false
    battery:
      voltage_warning: 3.8  # V
      voltage_critical: 3.7 # V
    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5
    #firmware_logging:
    #   enabled: true
    #   default_topics:
    #   pose:
    #     frequency: 1 # Hz
    #   custom_topics:
    #    topic_name3: 
    #      frequency: 1
    #      vars: ["acc.x", "acc.y", "acc.z"]

  cf21_mocap_deck:
    motion_capture:
      enabled: false
      # only if enabled; see motion_capture.yaml
      marker: mocap_deck
      dynamics: default
    big_quad: false
    battery:
      voltage_warning: 3.8  # V
      voltage_critical: 3.7 # V

    # firmware_params:
    #   kalman:
    #     pNAcc_xy: 1.0 # default 0.5

# global settings for all robots
all:
  # firmware logging for all drones (use robot_types/type_name to set per type, or
  # robots/drone_name to set per drone)
  firmware_logging:
    enabled: true
    default_topics:
     # remove to disable default topic
      pose:
        frequency: 10 # Hz
    #custom_topics:
    #  topic_name1:
    #    frequency: 10 # Hz
    #    vars: ["stateEstimateZ.x", "stateEstimateZ.y", "stateEstimateZ.z", "pm.vbat"]
    #  topic_name2:
    #    frequency: 1 # Hz
    #    vars: ["stabilizer.roll", "stabilizer.pitch", "stabilizer.yaw"]
  # firmware parameters for all drones (use robot_types/type_name to set per type, or
  # robots/drone_name to set per drone)
  firmware_params:
    commander:
      enHighLevel: 1
    stabilizer:
      estimator: 2 # 1: complementary, 2: kalman
      controller: 1 # 1: PID, 2: mellinger
    # ring:
    #   effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
    #   solidBlue: 255 # if set to solid color
    #   solidGreen: 0 # if set to solid color
    #   solidRed: 0 # if set to solid color
    #   headlightEnable: 0
    locSrv:
      extPosStdDev: 1e-3
      extQuatStdDev: 0.5e-1
    # kalman:
    #   resetEstimation: 1
  broadcasts:
    num_repeats: 15 # number of times broadcast commands are repeated
    delay_between_repeats_ms: 1 # delay in milliseconds between individual repeats`

2.motion_capture.yaml
`/motion_capture_tracking:
  ros__parameters:
    type: "optitrack"
    hostname: "192.168.0.80"

    mode: "motionCapture" # one of motionCapture,libRigidBodyTracker

    topics:
      poses:
        qos:
          mode: "sensor"
          deadline: 100.0 # Hz

    marker_configurations:
      default: # for standard Crazyflie
        offset: [0.0, 0.0, 0.0]
        points:
          p0: [   0.0,    0.0, 0.022]  # top
          p1: [-0.042,  0.042, 0.0  ]  # back left   (M3)
          p2: [-0.042, -0.042, 0.0  ]  # back right  (M2)
          p3: [ 0.042, -0.042, 0.0  ]  # front right (M1)
      default_single_marker:
        offset: [0.0, -0.01, -0.04]
        points:
          p0: [0.0177184,0.0139654,0.0557585]
      mocap_deck:
        offset: [0.0, 0.0, -0.01]
        points:
          p0: [0.03, 0.0, 0.0] # front
          p1: [0.00, -0.03, 0.0] # right
          p2: [-0.015, 0.0, 0.0] # back
          p3: [0.00, 0.03, 0.0] # left
      medium_frame:
        offset: [0.0, 0.0, -0.03]
        points:
          p0: [-0.00896228,-0.000716753,0.0716129]
          p1: [-0.0156318,0.0997402,0.0508162]
          p2: [0.0461693,-0.0881012,0.0380672]
          p3: [-0.0789959,-0.0269793,0.0461144]
      big_frame:
        offset: [0.0, 0.0, -0.06]
        points:
          p0: [0.0558163,-0.00196302,0.0945539]
          p1: [-0.0113941,0.00945842,0.0984811]
          p2: [-0.0306277,0.0514879,0.0520456]
          p3: [0.0535816,-0.0400775,0.0432799]

    dynamics_configurations:
      default:
        max_velocity: [2, 2, 3] # m/s
        max_angular_velocity: [20, 20, 10] # rad/s
        max_roll: 1.4 #rad
        max_pitch: 1.4 #rad
        max_fitness_score: 0.001

    # Rigid bodies will be automatically generated by the launch file
    # rigid_bodies:`

3.server.yaml
`/crazyflie_server:
  ros__parameters:
    warnings:
      frequency: 1.0 # report/run checks once per second
      motion_capture:
        warning_if_rate_outside: [80.0, 120.0]
    firmware_params:
      query_all_values_on_connect: False
    # simulation related
    sim:
      max_dt: 0 #0.1              # artificially limit the step() function (set to 0 to disable)
      backend: np                 # see backend folder for a list 
      visualizations:             # see visualization folder for a list
        rviz:
          enabled: true
        pdf:
          enabled: false
          output_file: "result.pdf"
        record_states:
          enabled: false
          output_dir: "state_info"
          logging_time: 0.2  # how many seconds to leave between logs
          file_formats: ["csv", "np"] # csv, np
        blender:
          enabled: false
          fps: 1           # frames per second
          cycle_bg: false  # if true, pictures will cycle through different environemt background images (useful for synthetic image generation). Otherwise a single environment background image will be used
          cf_cameras:      # names of crazyflies with cameras on them if enabled in `crazyflies.yaml`
            cf2:
              calibration:
                tvec: [0,0,0]
                rvec: [1.2092,-1.2092,1.2092]   # 0 deg tilt (cameras looks in front of crazyflie)
            cf5:
              calibration:
                tvec: [0,0,0]
                rvec: [ 0.61394313, -0.61394313,  1.48218982]   # 45 deg tilt
      controller: mellinger # none, pid, mellinger`

[image]

eunseo01, 2023-08-14:
After installing cpp, I was able to see the drone in rviz correctly. However, when I run 'ros2 run crazyflie_examples hello_world', the drone crashes during flight. I checked 'ros2 echo /cf2/pose' and the communication seems stable. I would like advice on what the problem might be. The YAML file is the same as the content I posted above.

closed 2023-08-14 by eunseo01

whoenig, 2023-08-14:
please open a new discussion with some more details.