IMRCLab/crazyswarm2 #279: Types of models supported by crazyswarm and their functions

gaurav713914, 2023-08-11:
May I know what different models does crazyflie support, for example a unicycle model, single integrator model, position control model etc. and especially where to find the commands for that? We want to implement unicycle, single integrator and double integrator models on a swarm of crazyflies.

whoenig, 2023-08-12:
Right now we have cmd_full_state, cmd_position, and cmd_hover. The firmware has a few more modes, see https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/crtp_commander_generic.c. cmd_full_state directly supports SI (set accelerations to 0), DI, and by using yaw also the unicycle (first or second order).

For others, I "emulate" the dynamics model on the host side and then use cmd_full_state (with position, velocity, and acceleration setpoints) or trajectories (polynomials) for execution.

gaurav713914, 2023-08-13:
How do we send only the velocities to the crazyflie? I mean we have all five arguments in the cmsFullState so we need to pass all 5 arguments, but if we need to implement the single integrator then do we have to set all the other arguments as 0 other than the velocity argument?

We want it to operate only with the velocity, we do not want to pass any target position, how do we do that, please help.

whoenig, 2023-08-14:
I assume you mean world-frame linear velocities? This is currently not exposed, but would be easy to add.

gaurav713914, 2023-08-14:
Yeah we mean world frame velocity for single integrator model because single integrator doesn't require Target position. Also, for unicycle we mean crazyflie frame velocity. How much time will it take to add this? 

Is there any way we could achieve single integrator model now using existing functions?

We're targeting two experimental research articles in domain of automatic control and for that we need unicycle model in one article and single integrator in other one.

whoenig, 2023-08-14:
Velocity control in body frame is the cmd_hover, see https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/crtp_commander_generic.c#L299-L307. 

For the one in world frame, it's already in crazyflie_cpp (see https://github.com/whoenig/crazyflie_cpp/blob/dev-coltrans/src/Crazyflie.cpp#L180-L186), so one would only need to add it in the server, similar to https://github.com/IMRCLab/crazyswarm2/blob/main/crazyflie/src/crazyflie_server.cpp#L913-L943. I don't think it's a lot of work, perhaps 1h + testing (which is the real work:-).

gaurav713914, 2023-08-15:
I tried using cmd_hover to publish the velocities in unicycle model but I am not able to do it and have posted the same on discussions page. I also tried adding world frame velocities to server but got a lot of errors, actually I am new to ROS and I was earlier working with flowdeck in a windows PC but now we are in process of researching on various models of robots in our newly setup Qualisys Motion Capture lab so these are the problems faced by us.

Can you please look into [this](https://github.com/IMRCLab/crazyswarm2/discussions/281) and also help with the world velocity thing. It would really help us in our research. I am a masters student so I am currently learning how to properly use ROS2.

gaurav713914, 2023-08-21:
We've edited the crazyflie_server.cpp and other dependent files to get cmd_hover subscription to work. Now, we're able to use cmd_hover without using the cflib backend. The following python code is how we're trying to get cmd_hover to work but there's an undesired drift in the y-axis when we try to have a simple yawrate of 10 with the linear velocities set to zero.

```
import  numpy as np
from crazyflie_py import Crazyswarm
import time
from crazyflie_interfaces.msg import Hover

import rclpy
from rclpy.node import Node
from geometry_msgs.msg import Twist
from crazyflie_interfaces.srv import Takeoff, Land

TAKEOFF_DURATION = 2.5
HOVER_DURATION = 10.0

class Uni(Node):
    def __init__(self):
        super().__init__('unicycle')
        self.declare_parameter('hover_height', 1.0)
        self.declare_parameter('robot_prefix', '/cf5')

        self.hover_height  = self.get_parameter('hover_height').value
        robot_prefix  = self.get_parameter('robot_prefix').value
        
        timer_period = 0.1
        self.timer = self.create_timer(timer_period, self.timer_callback)
        self.publisher_hover = self.create_publisher(Hover, robot_prefix + '/cmd_hover', 10)

    def timer_callback(self):
        msg = Hover()
        msg.vx = 0.0
        msg.vy = 0.0
        msg.yaw_rate = 10.0
        msg.z_distance = 1.0
        self.publisher_hover.publish(msg)

def main(args=None):
    swarm = Crazyswarm()
    timeHelper = swarm.timeHelper
    cf_1 = swarm.allcfs.crazyflies[0]
    cf_1.takeoff(targetHeight=1.0, duration=TAKEOFF_DURATION)
    timeHelper.sleep(TAKEOFF_DURATION)
    pos = np.array([0.0, 0.0, 1.0])
    cf_1.goTo(pos, 0, 5.0)
    timeHelper.sleep(5)
    k = 0
    
    dur = time.time() + 10
    
    msg = Hover()

    unicycle = Uni()
    
    rclpy.spin_once(unicycle)
        
    unicycle.destroy_node()
    rclpy.shutdown()

    cf_1.land(targetHeight=0.04, duration=2.5)

if __name__ == "__main__":
    main()
```
We plan to get VelocityWorldSetpoint working as well.

closed 2023-08-22 by gaurav713914