IMRCLab/crazyswarm2 #282: The 'hello world' flight is not working

eunseo01, 2023-08-17:
Hi, when I run 'ros2 run crazyflie_examples hello_world', the drone crashes during flight. I checked 'ros2 echo /cf2/pose' and the communication seems stable. I would like advice on what the problem might be. 
I will provide additional information here.

﻿crazyflies.yaml
[image]
[image]

﻿motion_capture.yaml
[image]
[image]

﻿server.yaml
[image]

video:
[Screencast from 08-17-2023 01_23_36 PM.webm](https://github.com/IMRCLab/crazyswarm2/assets/107001681/7c5b029f-5499-4b09-bcc1-83ed9c93b031)

https://github.com/IMRCLab/crazyswarm2/assets/107001681/db34770e-1a42-4e78-889d-a9614ede16a6

ewinge1, 2023-08-18:
Is the motion_cap_node warning you talking about the "ICP did not converge" warning in your screenshot? Are you getting this warning before you try to run hello world or after it crashes? Does the crazyflie show up as a stable transform (xyz axes) on Rviz when on the ground? If you line up the crazyflie to point (front is the point next to the power button) in the same direction as the positive-x direction in your space, do the crazyflie axes in Rviz align with the origin axes?

eunseo01, 2023-08-20:
Yes, 'ICP did not converge' is the motion_cap_node warning. And this warning appears right after starting rviz, before running hello world. When the crazyflie is on the ground, it's displayed as a stable transform in Rviz. Also, when you align the crazyflie in the same direction as its x-axis and the positive-x direction in your space, you confirmed that the axes in Rviz align with the origin axes.

ewinge1, 2023-08-21:
The behavior at the end of this error video is quite strange, it's as if the motion capture is jumping between where the crazyflie should be (landing) and where it actually is (crashed off screen). I had a hypothesis and was able to recreate that behavior, so it might apply to you. If you have both a real world server and a simulated server running at the same time, you'll get this jumping behavior. It also makes the crazyflie crash on takeoff and fail hello_world. Can you make sure you aren't running any old server processes? (be careful using ctrl+z to end things, ctrl+c or ctrl+\ are safer options)

If that wasn't the cause:
The crazyflie in Rviz should be more stable than what it appears in your video (and you shouldn't be getting the failed to converge errors). 

(I don't have experience with Optitrack and we use libRigidBodyTracker instead of motionCapture so I can only help so much here.)
It looks like you're running motion capture at 250Hz, which is quite high. Is that what your system is actually running at?  Turning down this frequency may solve some issues, too high a motion capture frequency may cause data latency issues (100Hz is what we and a few other groups use). 

I believe the name of the object you make to track in Optitrack must match the crazyflie name given in crazyflies.yaml. So you shouldn't be getting any messages about cf1 if you only have cf2 and cf5 in your yaml file. Again, the behavior at the end where it looks like it's jumping between the real position of the crazyflie and where it should be is quite odd.

eunseo01, 2023-08-22:
1. Wired LAN connection: It might be an internet problem, so we changed from wifi to the same network line as motion capture. But that wasn't how it was handled. 2. I saw the camera frame rate as 100hz, but the warning is the same. 3. Looking at the list of ros2 topics, there are 2 topics related to cf6 (other crazyflie). (We are using cf9 now.) Have you ever had this problem? Is there a need to reinstall ros2?
[image]
>> when we change Camera Frame Rate to 100Hz 

[image]
>> when we change Camera Frame Rate to 240Hz 

[image]
>> topic list

ewinge1, 2023-08-22:
The cf6 topics come from launch.py, which initializes cf6 to operate with a joystick or gamepad if one is available. You can remove that teleop node of the launch file, but it's not causing your issues.

The 100hz output looks better than the 240hz, as you have many less ICP failed to converge errors. Note that the capture rate warnings come once a second and are much less of a concerning warning than ICP failed to converge. The ICP issues are coming much more frequently at 240hz. Does it still crash at 100hz? A wired connection to the computer running Optitrack is more reliable than a wifi connection and will reduce latency and improve reliability.

eunseo01, 2023-08-23:
I can still see it crashing at 100hz.. Is there another solution? I'm using ros2 galactic right now, so can I change it to humble? Can you let me know if you need any more information?

ewinge1, 2023-08-25:
The mocap did not receive data does indeed mean that your mocap and crazyflie server are not communicating. In my experience, it usually means the ip or host name in the yaml file is wrong. You'll also see it if you pause tracking on accident (space bar in Vicon Tracker) or are in object tracking mode and have no objects selected.

If you aren't receiving mocap data, the crazyflie will always crash. There is no issue with using galactic and this is very likely not related to your ros2 install at all.