IMRCLab/crazyswarm2 #296: Ground Truth for Sim Backend

matthew-william-lock, 2023-10-05:
Hey all, for the simulation backend, I am wondering if there is a mechanism for getting the state of the UAV? For example, after stepping one could publish an Odometry message. Perhaps at a lower rate.

https://github.com/IMRCLab/crazyswarm2/blob/379601ea4c5f98f82c96749bfbe8e82a458ea9f9/crazyflie_sim/crazyflie_sim/backend/np.py#L37

If this does not exist and might be useful, I would be happy to create something simple and make a PR 😄

whoenig, 2023-10-05:
Where do you want to get/query the state? Each visualization has access to it, see https://github.com/IMRCLab/crazyswarm2/tree/main/crazyflie_sim/crazyflie_sim/visualization. For example, one can write the state(s) to a CSV file for later plotting. A new visualization could be written for publishing a ROS message, but then it would make most sense to use the /pose message that is also used in the real drone (if logging is enabled).

matthew-william-lock, 2023-10-09:
I am looking to subscribe to a ROS topic to mimic the MOCAP system/state estimation as per your second suggestion.  I have yet to perform experimentation on the real platform and am looking to test in sim first, hence why I am looking for the ground truth

whoenig, 2023-10-09:
In that case this would be simulating the motion_capture_ros package, which publishes on `/pose` as well as via tf. Another option would be `/<robotname>/pose`, which would simulate the behavior when using the on-board estimated state from the EKF rather than the global mocap state.