IMRCLab/crazyswarm2 #297: cmd_full_state example not working

matthew-william-lock, 2023-10-05:
Hey, when I run the ```cmd_full_state.py``` example I find that the drone takes off, but drops immediately when ```executeTrajectory``` is triggered.

## Steps to reproduce

In terminal 1
```
ros2 launch crazyflie launch.py backend:=sim
```
In terminal 2
```
 ros2 run crazyflie_examples cmd_full_state --ros-args -p use_sim_time:=True
```

whoenig, 2023-10-05:
This happens with all the default settings in the main branch, or do you have any changes in the config files?

matthew-william-lock, 2023-10-09:
This happens on the main branch without any changes, just a fresh clone

matthew-william-lock, 2023-10-09:
@whoenig I have found the reason for the behaviour 

https://github.com/IMRCLab/crazyswarm2/blob/379601ea4c5f98f82c96749bfbe8e82a458ea9f9/crazyflie_sim/crazyflie_sim/crazyflie_sil.py#L128

In other words, the simulation will only computer a control action when in ```CrazyflieSIL.MODE_HIGH_POLY``` mode, otherwise zero input is commanded. Is there perhaps a reason for this?

matthew-william-lock, 2023-10-09:
> @whoenig I have found the reason for the behaviour
>
> https://github.com/IMRCLab/crazyswarm2/blob/379601ea4c5f98f82c96749bfbe8e82a458ea9f9/crazyflie_sim/crazyflie_sim/crazyflie_sil.py#L284
>
> In other [...]

It seems to work when I remove this check

whoenig, 2023-10-09:
I think this should be `self.mode == CrazyflieSIL.MODE_IDLE`, basically to prevent it to fly off if no command is issued at all.

PR #362 (2023-11-16): sim: fix cmd_full_state example
Fixes #297
closed 2023-11-16 by whoenig