IMRCLab/crazyswarm2 #300: Slow updating of pose/scan topics with C++ backend

williamleong, 2023-10-10:
@liewyungjun and I tested the latest commit 379601ea and found that the pose and scan logging topics (at 10hz) were updated very slowly (more than a few seconds delay).

The problem seems to be introduced by the commit 56487ac3 for multithreading. When we roll back to 2e752ccb, it seems to perform as expected, where the pose and scan updates correctly in "real time" as we move the CF around.

I am not sure if it is entirely due to multithreading or the use of the `rmw_qos_profile_services_default` QOS profile.

whoenig, 2023-10-10:
Do you use a motion capture during your test as well? If so, at what frequency do you operate it?

Thanks for the report, especially knowing that this is a regression is very helpful!

williamleong, 2023-10-10:
Thanks for getting back. We didn't use a motion capture system.

PR #329 (2023-11-15): cpp backend: process all incoming radio messages
Process all radio messages every millisecond. The behavior
before was to process one message every 100 ms, which caused
long delays for incoming messages (such as log topics).

Fixes #300
commit 4bad8106 (2023-11-15): cpp backend: process all incoming radio messages

Process all radio messages every millisecond. The behavior
before was to process one message every 100 ms, which caused
long delays for incoming messages (such as log topics).

Fixes #300
closed 2023-11-15 by Khaledwahba1994