IMRCLab/crazyswarm2 #301: hello_worl: selected interface "lo" is not multicast-capable: disabling multicast

AlEXANDER9901, 2023-10-10:
"I want to run the command 'ros2 run crazyflie_examples hello_world --ros-args -p use_sim_time:=True,' but I encountered the following message related to multicast. Can I get some advice?

[image]
1696929214.140052 [0] hello_worl: selected interface "lo" is not multicast-capable: disabling multicast

ewinge1, 2023-10-10:
The instructions are a little unclear here, but you need to have the crazyflie_server running (i.e. ros2 launch crazyflie launch.py) when the param set commands are run. The node that you're missing is the crazyflie_server node, which can be started via launch.

ewinge1, 2023-10-10:
Also, was the title related to this issue or was it meant for something else?