IMRCLab/crazyswarm2 #303: ros2 param set

ntukenmez3, 2023-10-12:
What could be the reason for the following output, thanks.

```
ntukenmez3@ae-icps-407120:~/Documents/ros2_ws$ ros2 param set crazyflie_server cf1.params.commander.enHighLevel 1
Setting parameter failed
```

knmcguire, 2023-10-13:
Hi! We need more information but it seems that setting the parameter has failed as the error describes it.

 Is the crazyswarm server running in another terminal? Does that has an output or error when you try to do this?

ntukenmez3, 2023-10-13:
Yeap, I got the error. I run firstly the SIL simulation. It works. And then I tried the physical implementation part on the ros2-crazyswarm **Usage part**. I run firstly crazy server. Then the other part on that website.

ntukenmez3, 2023-10-13:
I added the related link here [cf2 ros2](https://imrclab.github.io/crazyswarm2/usage.html). I started to **Physical experiments part** to see how it works. But, it did not work. Simulation part works properly btw.

ntukenmez3, 2023-10-13:
> Hi! We need more information but it seems that setting the parameter has failed as the error describes it.
> 
> Is the crazyswarm server running in another terminal? Does that has an output or error when you try to do this?

How can I be sure that it works, thanks.

whoenig, 2023-10-13:
You need to run the crazyflie server in a second terminal `ros2 launch crazyflie launch.py` and report its output here.

ntukenmez3, 2023-10-13:
In the tutorial, that part is given after above commands, is that right, thanks

whoenig, 2023-10-13:
No, the tutorial is misleading there.

ntukenmez3, 2023-10-13:
**The terminal output:**

```
ntukenmez3@ae-icps-407120:~/Documents/ros2_ws/src$ ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E7E7 --mode sysoff
No Crazyradio with id=0 found.There are 0 Crazyradios connected.
[ros2run]: Process exited with failure 1
```

The cf2 is now on, but it could not find it.

whoenig, 2023-10-13:
for this first command the server should not be running. For the subsequent ones it should be running.

ntukenmez3, 2023-10-13:
the latter ones should be on the same terminal?

ntukenmez3, 2023-10-13:
Could you clarify a little bit more this code: `ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E7E7 --mode sysoff`, I run the codes as is following, it worked without the above code (`ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E7E7 --mode sysoff`).

1st: `ros2 launch crazyflie launch.py`

2nd: `ros2 param set crazyflie_server cf1.params.commander.enHighLevel 1`
3rd: `ros2 param set crazyflie_server cf1.params.stabilizer.estimator 2`

4th: `ros2 service call cf1/takeoff crazyflie_interfaces/srv/Takeoff "{height: 0.5, duration: {sec: 2}}"`
5th: `ros2 service call cf1/land crazyflie_interfaces/srv/Land "{height: 0.0, duration: {sec: 2}}"`

I would like to learn the function of `ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E7E7 --mode sysoff`, thanks.

ntukenmez3, 2023-10-16:
Correction on ros2-Crazyswarm2 tutorial:
I noticed that a **quotation mark (i.e., ")** is supposed to be added at the end of the following code, it is not a big deal but it might be pretty well if it is considered.

`ros2 service call /cf2/add_logging crazyflie_interfaces/srv/AddLogging "{topic_name: 'pose', frequency: 10}`   here + "

whoenig, 2023-10-16:
fixed - thanks.

knmcguire, 2023-10-18:
Just to clarify as this might not be clear for other users as well. There should be 2 terminals running, one with the server (`ros2 launch crazyflie launch.py`) and one where you run the parameters sets or services.

ntukenmez3, 2023-10-18:
> Could you clarify this code: `ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E7E7 --mode sysoff` a little bit more, I run the codes on the tutorial as is following, it worked without the above code (`ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E7E7 --mode sysoff`).
> 
> 1st: `ros2 launch crazyflie launch.py`
> 
> 2nd: `ros2 param set crazyflie_server cf1.params.commander.enHighLevel 1` 3rd: `ros2 param set crazyflie_server cf1.params.stabilizer.estimator 2`
> 
> 4th: `ros2 service call cf1/takeoff crazyflie_interfaces/srv/Takeoff "{height: 0.5, duration: {sec: 2}}"` 5th: `ros2 service call cf1/land crazyflie_interfaces/srv/Land "{height: 0.0, duration: {sec: 2}}"`
> 
> I would like to learn the function of `ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E7E7 --mode sysoff`, thanks.

Could you explain this question a little more?

ntukenmez3, 2023-10-18:
> No, the tutorial is misleading there.

I think that this part is supposed to be edited to make the tutorial looks functional, thanks.

ntukenmez3, 2023-10-18:
Lastly, is it needed to reboot the crazyflie by using `ros2 run crazyflie reboot --uri radio://0/80/2M/E7E7E7E7E7 --mode sysoff` ?

knmcguire, 2023-10-19:
No, it is not. But it is good practice to reboot the crazyflie before starting anything new. 

I'm closing this as we updated the doc to present it more clearly. If you have any more questions I'd suggest to start a discussion instead of an issue. Issues are usually pieces of work that needs to be fixed, and if it is unclear if it is a bug or just a misinterpretation of the tutorials, the discussions is the best place for that to start.

closed 2023-10-19 by knmcguire