IMRCLab/crazyswarm2 #307: ROS 2 Tutorials-Teleoperation keyboard

ntukenmez3, 2023-10-17:
I set up all configs and the related parts. Also, I've not gotten any errors but the cf is not getting the keyboard command for take-off. I added the terminal outputs associated with the related parts below. What could be the reason, thanks.

firs command:
`ros2 launch crazyflie_examples keyboard_velmux_launch.py`
**terminal output:**

```
ntukenmez3@ae-icps-407120:~/Documents/ros2_ws/src$ ros2 launch crazyflie_examples keyboard_velmux_launch.py 
[INFO] [launch]: All log files can be found below /home/ntukenmez3/.ros/log/2023-10-17-17-40-13-213907-ae-icps-407120-3879157
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [crazyflie_server.py-1]: process started with pid [3879158]
[INFO] [vel_mux.py-2]: process started with pid [3879160]
[crazyflie_server.py-1] [INFO] [1697578815.432539570] [crazyflie_server]:  radio://0/80/2M/E7E7E7E7E7 is fully connected!
[crazyflie_server.py-1] [INFO] [1697578815.432841106] [crazyflie_server]: All Crazyflies are fully connected!
[crazyflie_server.py-1] [INFO] [1697578815.437555995] [crazyflie_server]:  radio://0/80/2M/E7E7E7E7E7: commander.enHighLevel is set to 1
[crazyflie_server.py-1] [INFO] [1697578815.459489481] [crazyflie_server]:  radio://0/80/2M/E7E7E7E7E7: locSrv.extPosStdDev is set to 0.001
[crazyflie_server.py-1] [INFO] [1697578815.459974424] [crazyflie_server]:  radio://0/80/2M/E7E7E7E7E7: locSrv.extQuatStdDev is set to 0.05
[crazyflie_server.py-1] [INFO] [1697578815.476950645] [crazyflie_server]:  radio://0/80/2M/E7E7E7E7E7: stabilizer.controller is set to 1
[crazyflie_server.py-1] [INFO] [1697578815.477449967] [crazyflie_server]:  radio://0/80/2M/E7E7E7E7E7: stabilizer.estimator is set to 2
[crazyflie_server.py-1] [INFO] [1697578815.483605848] [crazyflie_server]: All Crazyflies parameters are initialized
[crazyflie_server.py-1] [INFO] [1697578815.485278574] [crazyflie_server]: All Crazyflies loggging are initialized
```
second command:

`ros2 run teleop_twist_keyboard telop_twist_keyboard`

**terminal output:**

```
ntukenmez3@ae-icps-407120:~/Documents/ros2_ws/src$ ros2 run teleop_twist_keyboard teleop_twist_keyboard 

This node takes keypresses from the keyboard and publishes them
as Twist messages. It works best with a US keyboard layout.
---------------------------
Moving around:
   u    i    o
   j    k    l
   m    ,    .

For Holonomic mode (strafing), hold down the shift key:
---------------------------
   U    I    O
   J    K    L
   M    <    >

t : up (+z)
b : down (-z)

anything else : stop

q/z : increase/decrease max speeds by 10%
w/x : increase/decrease only linear speed by 10%
e/c : increase/decrease only angular speed by 10%

CTRL-C to quit

currently:	speed 0.5	turn 1.0 

```

knmcguire, 2023-10-18:
Hi! I can confirm this! it seems that the vel_mux script is not starting up from the launch file so I'll need to check why that is.

knmcguire, 2023-10-18:
aha found it pretty quickly. You've got to change the crazyflie id here. 

https://github.com/IMRCLab/crazyswarm2/blob/a83684e31f7a164f3729f98fadc89e775293be19/crazyflie_examples/launch/keyboard_velmux_launch.py#L34

I did find, that if you take off and land, you can't take off again. So that is a different issue. For now you've got to restart the crazyflie.

ntukenmez3, 2023-10-18:
> aha found it pretty quickly. You've got to change the crazyflie id here. [...]

What could be the reason for this issue, I mean after the first take-off and land motion, why cannot send a take-off command?

ntukenmez3, 2023-10-18:
After the crazyflie id was changed as "`cf1`-->`c231`", it worked pretty well.

knmcguire, 2023-10-19:
The not taking off after a land, it's probably due to the chances in the latest crazyflie firmware so if you have an older release of the firmware it should work fine.

ntukenmez3, 2023-10-19:
I tried it, but it did not work.

knmcguire, 2023-10-19:
ahh oke then you have the latest release. No worries, I'm working on it :) https://github.com/IMRCLab/crazyswarm2/pull/309

with the current situation, you just have to restart the crazyflie everytime you want it to take off again...

knmcguire, 2023-10-19:
I'm converting this to a dicussion as this wasn't really an error but a support question. For the later issue there is already an issue made https://github.com/IMRCLab/crazyswarm2/issues/308

PR #309 (2023-11-16): cflib Add notify setpoints stop
adds the notify setpoint stop to crazyswarm2.

It should also fix some doc about the teleop_keyboard tutorial and the tutorial itself.

fixes #307 and #308 

This need some flight testing though before merge.
commit 12e602b9 (2023-11-16): fix explaination example. Fixes #307