IMRCLab/crazyswarm2 #323: Optitrack rigid_bodies example

matthew-william-lock, 2023-10-26:
Hey, I am trying to get the pose from the optitrack motion capture system. I have created a rigid body with the optitrack and am struggling get any data streaming to crazyswarm2. 

Is there anything similar to the config file in [mocap_optitrack](https://github.com/ros-drivers/mocap_optitrack/blob/foxy-devel/config/mocap.yaml) ?

Otherwise I am not sure what is missing... I can successfully query the command port, but do not see any data on the data_port.

Any suggestions are appreciated!

whoenig, 2023-10-26:
This usually means that IP and/or port are incorrect.

What version of Motive are you running?

If you are on Linux, you can recompile with the official SDK enabled (https://github.com/IMRCLab/libmotioncapture/blob/5e892e8a3402d1753a02c384738d6dd2c94c8417/CMakeLists.txt#L8), and then use "optitrack_closed_source" as type.

whoenig, 2023-10-26:
And the config is here: https://github.com/IMRCLab/crazyswarm2/blob/main/crazyflie/config/motion_capture.yaml#L3-L4. libmotioncapture is not related to mocap_optitrack and uses a completely different codebase.

matthew-william-lock, 2023-10-26:
Thank you for the quick support @whoenig !

We’re using motive 2.3 on a windows host machine. I have doubled checked that both the host machine and my ubuntu machine can ping the gateway, and can both ping each other.

In terms of recompiling the SDK, did you mean in the machine running motive ? Or would that help on my client machine as well ?

My apologies for the misunderstanding regarding the config, I was just wondering if rigid bodies had to be declared explicitly but I see not that this is handled in the launch file.

whoenig, 2023-10-26:
Motive 2.3 should work fine, although we use 3.0 as our daily driver.

You essentially change the one line I pointed you to (it will be in your ros2 workspace under     motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture) and rerun `colcon build` on the Linux machine. No changes are needed in Windows/Motive (assuming you have NatNet Streaming enabled in Motive already).

matthew-william-lock, 2023-10-27:
Great, will give it a go! Just to confirm, I should change the line to 
``` option(LIBMOTIONCAPTURE_ENABLE_OPTITRACK_CLOSED_SOURCE "Enable Optitrack (Closed Source)" ON) ``` ?

matthew-william-lock, 2023-10-27:
> Great, will give it a go! Just to confirm, I should change the line to `option(LIBMOTIONCAPTURE_ENABLE_OPTITRACK_CLOSED_SOURCE "Enable Optitrack (Closed Source)" ON)` ?

This unfortunately does not help. Streaming has been enabled on motive as shown in  https://docs.optitrack.com/v/v2.3/motive/data-streaming.

matthew-william-lock, 2023-10-27:
I have quickly tried [ros2-mocap_optitrack](https://github.com/tud-cor-sr/ros2-mocap_optitrack) which successfully streams the motion capture data, so the network/optitrack setup should be fine :sweat_smile:

matthew-william-lock, 2023-10-27:
> Motive 2.3 should work fine, although we use 3.0 as our daily driver.
>
> You essentially change the one line I pointed you to (it will be in your ros2 workspace under [...]

I forgot to change the type in the launch file, but I get this error now when trying to launch

```
[INFO] [motion_capture_tracking_node-1]: process started with pid [82067]
[motion_capture_tracking_node-1] /home/matt/ros2_ws/install/motion_capture_tracking/lib/motion_capture_tracking/motion_capture_tracking_node: error while loading shared libraries: libNatNet.so: cannot open shared object file: No such file or directory
[ERROR] [motion_capture_tracking_node-1]: process has died [pid 82067, exit code 127, cmd '/home/matt/ros2_ws/install/motion_capture_tracking/lib/motion_capture_tracking/motion_capture_tracking_node --ros-args -r __node:=motion_capture_tracking --params-file /tmp/launch_params_nrlw38w1'].
```

Should I install libNatNet?

whoenig, 2023-10-27:
No separate library needed. Try setting up https://github.com/whoenig/NatNetSDKCrossplatform/tree/master and see if that works for you (the closed source version).

matthew-william-lock, 2023-10-27:
The closed-source version works!

> No separate library needed. Try setting up https://github.com/whoenig/NatNetSDKCrossplatform/tree/master and see if that works for you (the closed source version).

I still get the ```libNatNet.so: cannot open shared object file: No such file or directory``` error when launching though

whoenig, 2023-10-27:
Strange, but also not well tested... - you can use `ldd <path/to/motion_capture_tracking_node>` to figure out where it is looking for it. It's part of NatNetSDKCrossplatform.

matthew-william-lock, 2023-10-27:
Sure, I get the following (truncated)

```
ldd /home/matt/ros2_ws/install/motion_capture_tracking/lib/motion_capture_tracking/motion_capture_tracking_node
	linux-vdso.so.1 (0x00007ffd801a6000)
	libmotion_capture_tracking_interfaces__rosidl_typesupport_cpp.so => /home/matt/ros2_ws/install/motion_capture_tracking_interfaces/lib/libmotion_capture_tracking_interfaces__rosidl_typesupport_cpp.so (0x00007f662279b000)
	libpcl_search.so.1.12 => /lib/x86_64-linux-gnu/libpcl_search.so.1.12 (0x00007f6622019000)
	libpcl_kdtree.so.1.12 => /lib/x86_64-linux-gnu/libpcl_kdtree.so.1.12 (0x00007f6622662000)
	libpcl_common.so.1.12 => /lib/x86_64-linux-gnu/libpcl_common.so.1.12 (0x00007f6621f7f000)
	libboost_thread.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_thread.so.1.74.0 (0x00007f6621f5d000)
	libNatNet.so => not found
```

matthew-william-lock, 2023-10-27:
I found the ```libNatNet.so``` and added

```export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/matt/ros2_ws/src/motion_capture_tracking/motion_capture_tracking/deps/libmotioncapture/deps/NatNetSDKCrossplatform/lib/ubuntu``` to my bashrc.

Everything seems to work now :) Might be worth further investigation for other though who face the same issue

matthew-william-lock, 2023-10-27:
One more question if I may, how does crazyswarm handle the motion capture information? For example, will it take preference for what is published on the ```/cf1/pose``` topic ? I am a little new to the crazyswarm environment :sweat_smile:

whoenig, 2023-10-27:
- motion_capture_tracking publishes on `/poses` and via TF
- crazyswarm_server is broadcasting this information to the Crazyflie using the Crazyradio
- `cf<x>/pose` (if enabled) is the state estimate that comes from the on-board EKF (fusing mocap + IMU)

closed 2023-11-01 by whoenig