IMRCLab/crazyswarm2 #327: server cpp bug helloworld latest master crazyflie-firmware

knmcguire, 2023-10-31:
Hi! 

During release testing I noticed that the hello world example of Crazyswarm2 is not working anymore.

**How to recreate**
1. Flash Crazyflie with the latest master or the 2023.10-rc1 release: https://github.com/bitcraze/crazyflie-firmware
2.  Run the cpp link backend server with 
```ros2 launch crazyflie launch.py```
3. Run helloworld with
``` ros2 run crazyflie_examples world```

**symptoms**
* Crazyflie stays grounded, no response
* The hello_world script freezes
* No error on the server side, all seem normal

**Additional info**
* Cflib backend seem to work (`backend:=cflib`) but that one has some issues of itself as well: https://github.com/IMRCLab/crazyswarm2/issues/308. Maybe related?

knmcguire, 2023-11-02:
The plot thickens!

So when I just flashed the lastest crazyflie-firmware on a random crazyflie, it worked... just to test my own sanity I did the following tests with cpp backend + helloworld

- 2023.07 - works
- then flash with master crazyflie-firmware (leaving the nrf the same) - worked
- 2023.10-rc1 (latest master firmware of both nrf and stm - freeze

All of these still worked on the cflib backend.

But... then it might be something in the nrf somehow? Anyway I've got the zip file here  that fails (I removed the candidate release since I don't want it to turn on every bodies cf client)

[firmware-cf2-2023.10-rc1.zip](https://github.com/IMRCLab/crazyswarm2/files/13238877/firmware-cf2-2023.10-rc1.zip)

knmcguire, 2023-11-02:
So the only PR that is between some releases ago and  the NRF is this one https://github.com/bitcraze/crazyflie2-nrf-firmware/pull/91... can't see what  should be causing it though

knmcguire, 2023-11-02:
Alright, another development that brings us closer hopefully:

I've now moved to the nrf firmware for debugging. If I make the standard CF2 (with BLE and S110 still on):
```
make BLE=1 S110=1 -j12
```
take off with cpp backend freezes

If I disable BLE and S110:
```
make BLE=0 S110=0 -j12
```
It works! So it must be something going on in those if statements

krichardsson, 2023-11-02:
I found a difference in the NFR, try https://github.com/bitcraze/crazyflie2-nrf-firmware/pull/92 to see if it fixes the problem

knmcguire, 2023-11-03:
thanks @krichardsson ! I can test it out shortly next week or after the 12th

knmcguire, 2023-11-14:
just testflown it and it works again! thanks @krichardsson

closed 2023-11-14 by knmcguire