IMRCLab/crazyswarm2 #360: Error running the ROS 2/Nav2 Bringup tutorial

williamleong, 2023-11-16:
Had a couple of errors show up while running the tutorial, but I don't know if it's specific to my setup:

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): [Errno 2] No such file or directory: '/xxx/ros2_ws/install/crazyflie_examples/share/crazyflie_examples/nav2_params.yaml'

[ERROR] [launch]: Caught exception in launch (see debug for traceback): name 'true' is not defined
```
Modifying `multiranger_nav2_launch.py` fixed them:

```
IncludeLaunchDescription(
    PythonLaunchDescriptionSource(
        os.path.join(bringup_launch_dir, 'bringup_launch.py')),
    launch_arguments={
        'slam': 'False',
        'use_sim_time': 'false',
        'map': cf_examples_dir + '/data/' + map_name + '.yaml',
        'params_file': os.path.join(cf_examples_dir, 'config/nav2_params.yaml'),
        'autostart': 'true',
        'use_composition': 'True',
        'transform_publish_period': '0.02'
        }.items()
),
```

williamleong, 2023-11-16:
I'm guessing the other settings 'false' and 'true' needs to be 'False' and 'True' as well. Have tried running it that way and it seems to work okay.

knmcguire, 2023-11-16:
hmm intersting, this usually worked out fine. so maybe ROS 2 has had some updates.

Are you currently on the latest humble?

knmcguire, 2023-11-16:
I'm able to confirm this. I think this is due to fixes to the NAV2 bringup package. In particular the `use_composition` fails when it is lowercase true or false. 

I'll submit a PR to fix this.

closed 2023-11-16 by knmcguire

williamleong, 2023-11-17:
Thanks and apologies, I wasn't able to clarify earlier due to time zone differences :)

PR #365 (2023-11-16): NAV2 example fix uppercase error and lost file
fixes #360