IMRCLab/crazyswarm2 #376: Warnings while doing "colcon build" (Beginner)

kusmauld, 2023-11-27:
I'm currently attempting to setup a simulation on my computer for the crazyflie. I've been running into multiple problems, but I believe these errors/warnings are potentially causing the problem.
Program Versions
- Ubuntu 22.04
- ROS Humble
- Python 3.10.12
- pip 23.3.1

Output of colcon build --symlink-install
Starting >>> crazyflie_interfaces
Starting >>> motion_capture_tracking_interfaces
Finished <<< motion_capture_tracking_interfaces [1.47s]                       
Starting >>> motion_capture_tracking
Finished <<< crazyflie_interfaces [2.01s]                                     
Starting >>> crazyflie_py
Starting >>> crazyflie
Starting >>> crazyflie_sim
--- stderr: crazyflie_py                                                                          
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< crazyflie_py [0.96s]
Starting >>> crazyflie_examples
--- stderr: crazyflie_sim                                                   
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< crazyflie_sim [1.56s]
Finished <<< crazyflie [1.64s]
--- stderr: crazyflie_examples                                              
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< crazyflie_examples [1.25s]
--- stderr: motion_capture_tracking                             
CMake Warning (dev) at deps/librigidbodytracker/CMakeLists.txt:7 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  CMake variable PCL_ROOT is set to:

    /usr

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

** WARNING ** io features related to pcap will be disabled
CMake Deprecation Warning at deps/libmotioncapture/deps/vrpn/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at deps/libmotioncapture/deps/vrpn/quat/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at deps/libmotioncapture/deps/vrpn/cmake/FindOpenHaptics.cmake:56 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  deps/libmotioncapture/deps/vrpn/CMakeLists.txt:468 (find_package)

CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (I2CDEV)
  does not match the name of the calling package (libi2c-dev).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  deps/libmotioncapture/deps/vrpn/cmake/Findlibi2c-dev.cmake:43 (find_package_handle_standard_args)
  deps/libmotioncapture/deps/vrpn/CMakeLists.txt:518 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at deps/libmotioncapture/deps/vrpn/client_src/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

---
Finished <<< motion_capture_tracking [3.27s]

Summary: 7 packages finished [5.31s]
  4 packages had stderr output: crazyflie_examples crazyflie_py crazyflie_sim motion_capture_tracking

Eric-Ewing, 2023-11-27:
What are the problems you are having? These warnings are primarily about setuptools being deprecated, which shouldn't actually cause any fatal issues. Your packages built successfully.

I believe if you run `colcon build` again, it will say it built without any issues.

kusmauld, 2023-11-27:
My problem has been fixed. I was attempting to run the simulation, but the errors I was getting was "cffirmware module not found". I found out that the reason was because after building the python bindings, when I did "export PYTHONPATH=<replace-with-path-to>/crazyflie-firmware/build:$PYTHONPATH", I used a ~ to refer to my user's home folder (~crazyfile-firmware/build), but was suppose to type out the full path (/home/{username}/crazyfile-firmware/build). I thought that those errors would've been a problem but appear to be fine. Thanks for your response!

closed 2023-11-27 by kusmauld