IMRCLab/crazyswarm2 #405: Simulation - investigate why such small dts are needed

whoenig, 2024-01-18:
Crazyflies crash if the dt is increased from 0.0005, even though in other simulations with the Python bindings and the same physics backend we achieve 0.01s.

whoenig, 2024-01-24:
This might be a control issue in combination with motor saturation. The following investigations have been done:

- Tried different physics backends (same outcome)
- Tried fixing the firmware ticks issue directly in the firmware (similar outcome, although behavior does change in this case)
- Tried the Lee controller, for which we had successful simulations at higher dt's. This works reasonably well. Here we tried two versions: one that uses the polynomials to convert force/torque -> PWM -> RPM -> Force, and one that uses the force directly. Both worked fine, although avoiding the multiple conversions had a lower tracking error.

Moving forward, the goal is to get the Lee controller in the official firmware, and use it as default with a higher dt in simulation.

jpreiss, 2024-03-20:
I have also been finding some weird stability issues testing with the Mellinger controller even with dt small. I noticed that the controller's roll/pitch moment commands were saturating a lot, even in benign conditions. I traced it to the `kd_omega_rp` term in the Mellinger controller. Setting `kd_omega_rp` to zero improves stability a lot for me. Curious if it also helps your issue?