IMRCLab/crazyswarm2 #423: crazyflie_server cflib - status - support for unicast/broadcast vars

whoenig, 2024-02-09:
The cflib status is currently missing some variables, since they are not in the latest release. Either wait for the next release, or add them conditionally (similar to the cpp backend).

NageshEranki, 2025-03-27:
Hi,

I am interested in expanding broadcasting capabilities to cflib. I successfully used `crazyflie-link-cpp` to send position set points and motion capture feedback for 5 drones. With a little help, I would love to contribute to this feature. 

I found this [ancient thread](https://forum.bitcraze.io/viewtopic.php?f=19&t=4807) that discusses this and was wondering how I could proceed from there.

knmcguire, 2025-03-30:
@whoenig just to clarify here, you are talking about the communciation quality values here, not the broadcasting functionality of cflib itself? I guess we could implement some of the link statistics now? https://www.bitcraze.io/documentation/repository/crazyflie-lib-python/master/api/cflib/crazyflie/link_statistics/

@NageshEranki since broadcasting is not yet part of the CFLib, this is a crazyflie specific feature request.  You can either ask the folks at Bitcraze what the current status is on the implementation on their discussions: https://discussions.bitcraze.io/, or ask on this issue: https://github.com/bitcraze/crazyflie-lib-python/issues/472. But I've seen that you have found that one already ;) (I started that issue but I do not work at Bitcraze anymore)

whoenig, 2025-03-30:
Yes, this was just for the status message, not for adding full broadcast capabilities.

knmcguire, 2025-07-11:
Let's close this. #686 already has fixed this and broadcasting is not yet included in the cflib anyway but that wasn't the point of this issue. 

Once this one is resolved let's open up another issue to add those stats too: https://github.com/bitcraze/crazyflie-lib-python/issues/472

closed 2025-07-11 by knmcguire