IMRCLab/crazyswarm2 #430: crazyflie.yaml -  launch accept other file

knmcguire, 2024-02-20:
For the efforts of intergrating crazyswarm2 into our testlab (https://github.com/bitcraze/crazyflie-testing/issues/25), it would be handy for the standard launch file to accept another crazyflie.yaml file as input.  Currently this is a bit hardcoded:

https://github.com/IMRCLab/crazyswarm2/blob/32e87d6f84a532bb23324b7e7633a0081fcbc026/crazyflie/launch/launch.py#L14-L17

Might be useful for others as well?

knmcguire, 2024-02-28:
So I've tried to take a look at this yesterday but I was a bit doubtful on the approach we should take. The launch file does quite some things to merge the content of 3 yaml files together, the crazyflie.yaml, the server.yaml and the motion_capture_yaml. Do we want to make all 3 these yaml file inserts possible? And all of them separately or do we want the user to make one big external file?

whoenig, 2024-02-28:
I was wondering that, too. I think it should be consistent with the overall architecture. If we decide to merge these 3 files (e.g., into a "cfg.yaml") then we only need the include to support that. If we keep the current architecture, we should be able to change the filename (and or packagename) of all three of them.

I think having a single cfg.yaml might be nicer, also because it makes submitting issues easier (one file, rather than us always asking how the other files look like).

knmcguire, 2024-02-28:
So if I understand correctly what you are proposing:
- Combine those 3 yamls together into cfg.yaml
- Ability to asign another external cfg.yaml through declarelaunchargument or something similar

How about the other yaml files in the [/config](https://github.com/IMRCLab/crazyswarm2/tree/main/crazyflie/config) folder though... those seems a bit more specialized, but I guess that the motion capture yaml file might be too specialized as well

whoenig, 2024-02-29:
Correct. I think the two options are:

1. Combine all yaml files into one and allow the user to specify an alternative *filename*.
2. Allow the user to specify an alternative foldername (or packagename). Optionally, we could dynamically use the default files if a particular file is missing in the given folder/package.

2 might be a bit cleaner, but has the disadvantage that the configuration is spread across many different files. Also the dynamic check can lead to fun bugs (e.g., if one misspells the folder name, the default files will be silently used).

knmcguire, 2024-03-07:
I do like the folder idea but I'm afraid that we'll need to work with 2 launch files still. I've tried to have an hand in PR #448 but that fails since the launch argument can not be used when the launch file is run.  The python style launch file was supposed to make it easier to do some 'fancy things' but looks can be deceiving as it is still quite limiting :sweat_smile: 

I can see if I can make a minimal example more close to this perhaps: https://docs.ros.org/en/galactic/Tutorials/Intermediate/Launch/Using-Substitutions.html Perhaps using that example launch file you made recently

Option 1 is the easiest to implement, but the most distributive to the crazyswarm2 user base... so I'm a bit afraid of going down that path.

knmcguire, 2024-03-10:
I've looked at the pathjoinsubtitution but I can't really find a good way to solve this for us. We want to a bit too much with our launch python script that python launch files was not completely made for.

How about we have a simple node that concatenate the yaml files instead of doing all the logic the launch file? We might be perhaps be able to do something there. Or perhaps in a way that NAV2 has done with rewriting the yaml file https://github.com/ros-planning/navigation2/blob/main/nav2_common/nav2_common/launch/rewritten_yaml.py. I got that idea from this ROS answers https://answers.ros.org/question/402129/ros2-substituting-a-parameter-from-confg-file-while-keeping-others-within-python-launch-file/

knmcguire, 2024-03-18:
Got a new tryout of a simpler way. https://github.com/IMRCLab/crazyswarm2/pull/454

closed 2024-04-11 by knmcguire